FreddyVRetro/ISA-PicoMEM #3: Boot hangs when 4096MB partitioned unformatted image is used

megatog615, 2024-05-17:
I created a new disk image in the Picomem setup screen, 4096MB (which settled on 4095MB), booted a win98 startup floppy and used parted to create a partition the full disk size. After a reboot, it hangs when booting to a floppy image using the system's boot code as well as the Picomem's boot code.

NOTE: I've gotten it to boot by doing exactly the same steps but setting it to 2048MB.

FreddyVRetro, 2024-05-17:
Thanks for the info.
4096Mb may be a bit too much, I never tested it.
Use Big partition is not recommended, it will be slow, the PicoMEM is yet not optimized for it (don't forget it is an 8Bit board)
(At minimum, use lot of disk cache)
Nice that you tries on Windows 98 :)

megatog615, 2024-05-17:
Well, I didn't try win98 really just the boot disk. I'm going to try Win95 actually.

megatog615, 2024-05-17:
The system seems to be operate fast enough even on 8-bit ISA card. DIR command completes very quickly in DOS.

closed 2024-10-11 by FreddyVRetro