FreddyVRetro/ISA-PicoMEM #4: IBM 5150 does not boot HDD from PicoMEM

tingtron, 2024-05-21:
I have a couple of HDD images (in RAW / flat format): one 20 MB VMDK from VirtualBox and one 40 MB IMG created with 86Box.

The issue is that they don't boot in PicoMEM - it just goes straight to ROM BASIC.
Yet both have PC-DOS 3.30 installed, and they boot in their respective emulators. 

I can select them in PicoMEM settings, and when booting from floppy, I can access the HDD files, run programs etc.

Also the 40 MB drive has a Primary C: and Extended D: Partitions. But the Logical drive on the Extended partition D: is not shown by FDISK from PC-DOS 3.30. Sometimes the root directory on D: would show garbage. I try to avoid writing anything to drive D: in PicoMEM. And wanted to confirm that Logical partitions are supported.

I know what PicoMEM examples use MS-DOS 3.31. But I wanted to go with the more authentic PC-DOS, which I believe only has up to version 3.30 (so partitions are up to 32 BM). Is such scenario supported?

tingtron, 2024-05-21:
I tested the 40 MB drive created by 86Box in DOSBox, and it booted fine and correctly showed the Logical partition D:.

I did have to specify the geometry "size" parameter in DOSBox (S,H,C = `17,8,615`), because 86Box uses very specific geometries for different drive sizes. (I guess they are trying to replicate real drives).

```
imgmount c "PCDOS33_xtide.img" -size 512,17,8,615
boot -l c
```

tingtron, 2024-05-21:
I didn't change or investigate the IRQ settings on PicoMEM or other hardware.
I am not sure if IRQ has anything to do with booting.

tingtron, 2024-05-22:
Also strange that PicoMEM changed the size of the 40 MB IMG above (PCDOS33_xtide.img) from 42,823,680 byte to 157,992,448 bytes (a 115168768 bytes increase).
So it might be because PicoMEM is confused by the geometry.

tingtron, 2024-05-22:
In the meantime, I was able to create a 30 MB disk inside PicoMEM settings, which has CHS: 60,16,63.
And after formatting with system option, it did become bootable.

This assumption (heads=16, sectors=63) is in line with how [DOSBox IMGMOUNT](https://www.dosbox.com/wiki/IMGMOUNT) interprets disk images: always as `-size 512,63,16,X` where heads and sectors are fixed and only cylinders are variable.

Note: this PicoMEM created disk image also boots in DOSBox without specifying the `-size` parameter.

FreddyVRetro, 2024-05-22:
Hi,
I am really sorry for that, I was sure I wrote an article/Explained how it work somewhere, but no.

The PicoMEM use the Geometry like DOSBOX.
So, like you finally found, when you create an image from 86Box or PCEM, You need to fix the geometry with 16 Heads and 63 Sectors.

Find the Geometry from a .IMG file is not simple.

You can also create the image from the PicoMEM BIOS Directly and format it from a DOS Image disk, like you can do it with 86BOX/PCEM

tingtron, 2024-05-22:
@FreddyVRetro No problem! Thank you for confirming what I discovered. I should've started by creating a disk image from PicoMEM. And thanks to @misterblack1's video I realized that.

Interestingly, 86Box disk image creation tool has a calculator for custom sizes. And the 16 Heads / 63 Sectors start after around 300 MB. And I am guessing this reflects some real disks heuristics and the limitations of the CHS geometry formulas. 

[image]

closed 2024-05-22 by tingtron