FreddyVRetro/ISA-PicoMEM #7: Compatibility list?

Ripper912, 2024-05-29:
Hi, is there a compatibility list ? Or did anybody already used with 286 ISA Toshiba T3200 notebook? I have one at home so can confirm/denial funcitionality.

FreddyVRetro, 2024-05-30:
Hi,
A minimal list can be found on this GitHub home readme file.

elbosito, 2024-06-02:
Hi , bought the card on Occc meet in Martigny , tried to run in my Amiga Sidecar but only see PicoMem Bios ....init:...End , no bios shows up , any hints

FreddyVRetro, 2024-06-02:
Hi,
If you have this it is a Good news.

Tes PicoMEM initialisation is in 2 Phases : What you saw is the Optionnal BOOT ROM initialisation and it was Ok.
Then, the PicoMEM BIOS redirect the Boot Strap interrupt to start the Setup.
If it does not start, it means that you have another board redirecting the Boot Strap interrupt After the PicoMEM BIOS.
Do you have an XTIDE or something else ?
If yes, the PicoMEM ROM should be placed after it. (in the config.txt file)

To force the PicoMEM Setup to start during the Optionnal ROM init, add PREBOOT in the config.txt file
https://github.com/FreddyVRetro/ISA-PicoMEM/tree/main/images

FreddyVRetro, 2024-07-11:
> Hi , bought the card on Occc meet in Martigny , tried to run in my Amiga Sidecar but only see PicoMem Bios ....init:...End , no bios shows up , any hints

Hi,
Did you try the PREBOOT option ?

FreddyVRetro, 2024-07-22:
Hi,

The last PicoMEM Fast RAM version Firmware may resolve part of the problems here, can you test ?

HarveyBrookes, 2024-08-03:
> Hi, is there a compatibility list ? Or did anybody already used with 286 ISA Toshiba T3200 notebook? I have one at home so can confirm/denial funcitionality.

I have it in a t3200e and only works if you slow the CPU down (red power light not green) by holding fn and page down. 
Just after mem test. 
If later than that the picomem fails with garbage on screen.  

Otherwise it's excellent. 
I have managed to make some basic hdd and floppy files which work flawlessly. 

I am split between XTIDE and Pico mem due to the options picomem offers such as mouse and network. But so far no joy at the default full 12 MHz mode only 6MHz (if I can find a way to change the ISA bus speed it might work properly, nothing obvious in the T3200e BIOS).

So in summary - yes it works but you have to slow the pc down to allow picomem to work. Do that and there's no issues.

Ripper912, 2024-08-12:
Hi HarveyBrookes tank you for your comment. Today i finally recieved my piece v 1.
14 with the latest July firmware. Can confirm the PicoMem works fine now on Toshiba T3200. Testet with USB hub and USB mouse also Adblib works ;-)

Great Job!
[image]

Ripper912, 2024-08-26:
Hi, have tested on Toshiba T3200 (286 notebook) the card is working (FDD,HDD, Adlib, Network...) except the RAM.By booting only "garbage" on the screen.(According Harvey downlscaling to 8MHz should help) however T3200 without e or SX doesnt support it.

FreddyVRetro, 2024-08-26:
Hi,

So, it seems you are 2 with T3200 and Garbage on screen only when RAM emulation is enabled ?
Then, other fonction work as normal ? It can boot ?

Garbage on screen may be due to the PicoMEM emulating RAM on the video buffer address, due to RAM address translation on the machine.

Ripper912, 2024-08-26:
Hi Freddy yes exactly. FDD,HDD emu works Adlib music asi well. Wifi is detected (havent tried). However if i try to boost with PicoMEM.exe in config sys it starts to put only random garbadge on the screen & doesnt boot anymore.(Stock clock of T3200 is 12MHz ) & have not yet found a way do downlscale ;( however your product is awsome

FreddyVRetro, 2024-08-26:
Hi,

Picomem.exe is a diagnostic software that read/write data on fixed address.
It should not really be used, it does not initialize anything.

To initialize the Keyboard shortcut, it is pminit /t

Station2381-sys, 2024-08-27:
[image]
[image]
[image]
[image]
[image]
[image]

Hi Freddy. I will also join with experience. Sorry if I missed something. I also have a Toshiba T3200. Adlib, wifi, USB, HDD, FDD, everything is OK. I directly tried connecting to the BBS via the wifi via the MTCP network settings. The only problem I have is with memory. It is not possible to turn on the EMS. Sector E is filled with "R" symbol and I don't know how to free this or where I am doing wrong. I have everything else set to OFF. I tried picomem on 286 and pentium 1 and same problem. The only one where I managed to turn on EMS was 386DX/40. See attached photos. 
Thank you for your work and time. PicoMem is great. :)

FreddyVRetro, 2024-10-04:
Hi,

Sorry I see your message only now.

You can move the PicoMEM BIOS to C800 to free the segment D and add EMS there

FreddyVRetro, 2024-10-07:
> Hi FreddyV,
>
> Sorry but how to do that please?(Move BIOS to C800)?
> Thanks for your respond

Hi,
Create a config.txt file and add this :
BIOS C800

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

Ripper912, 2024-10-08:
Hi FreddyV,

Thanks a lot. Finally I can confirm that by adding config.sys (BIOS C800)into root of the SD its possible to enable EMS under PicoMem setup & than when added under config.sys (for example Device=C:\Driver\Pmemm.exe) the additional RAM is also working under Toshiba T3200. So now finally everything (HDD,FDD,RAM, Sound, Wi-Fi) works - great great product ;)
[image]
[image]

closed 2024-10-08 by Ripper912