FreddyVRetro/ISA-PicoMEM #10: Memory mapping settings in May_28 firmware lost after power cycle.

johnwiggins, 2024-06-02:
Running on a Panasonic Senior Partner. PicoMEM is the first card that successfully brings this system to 640K!

However, interesting issue: May_28 firmware doesn't seem to keep memory mapping settings after power cycle. HDD and Wifi interrupt and address settings do persist, but not RAM settings.

Rolling back to April_21 firmware fixes the issue.

Brilliant piece of work, thank you!

FreddyVRetro, 2024-06-03:
Hi,
I will check if I can reproduce it.

The Settings are saved on the microSD every time the setup is RUN
If it fail, there is an error message displayed, did you see it ?

johnwiggins, 2024-06-03:
No, I did not see an error message.

The BIOS banner displays normally with a prompt to hit S for settings. When the timer expires, the banner showing the disk/memory configuration displays with the correct disk settings, but it reports that there is no RAM configured.

Thank you

FreddyVRetro, 2024-06-12:
Hi,
I found why and it will be corrected in next firmware.

If you create a config.txt file on the uSD, the EMS Settings will be saved.

FreddyVRetro, 2024-06-27:
Will be corrected in next firmware.

closed 2024-06-27 by FreddyVRetro

Gr8Britton, 2024-07-04:
> Hi, I found why and it will be corrected in next firmware.
> 
> If you create a config.txt file on the uSD, the EMS Settings will be saved.

Should this config.txt file be blank?  Also, should it be on the root of the uSD card?