FreddyVRetro/ISA-PicoMEM #12: WiFi bug in may28 firmware?

troydtaylor, 2024-06-11:
When using the April 24th firmware, I am able to connect to my wifi and get an IP from DHCP.  After upgrading to May 28th firmware, it does not work.  Only thing that was changed was the firmware.  Signal is about -63 to -65db.  While I know that is marginal, it works every time with the April 24th firmware.  See attached images.
[image]
[image]

troydtaylor, 2024-06-11:
Not sure if it matters but I am using an Orbi mesh network.

troydtaylor, 2024-06-17:
So it appears that the new firmware makes the antenna less sensitive and less able to communicate with marginal signals.  I put in a new Orbi on the workbench and the new may 28th firmware was able to connect.  Signal strength went from -63db to -38db.  Not sure why the old firmware works better with marginal signal strength than the new firmware, but that does appear to be the case.

FreddyVRetro, 2024-07-16:
Hi,
July 16 firmware I just released went back to some of the April 24 Wifi code, maybe you can try now ?

megatog615, 2024-07-18:
Not sure if it's related but in my case any time it actually connects the signal strength is "0db" which seems in error.

FreddyVRetro, 2024-07-22:
Hi,

I confirm, 0db means it fail to connect.

Latest firmware may help for all the problem found here, as I came mach to the previous firmware Wifi config.

troydtaylor, 2024-07-22:
Okay.  Thanks Freddy.  I will give it a try.

FreddyVRetro, 2024-10-11:
Hi @troydtaylor 

Since the firmware changed a lot, did you try the latest one ?

FreddyVRetro, 2025-05-13:
Many fix done on wifi side, maybe working now