FreddyVRetro/ISA-PicoMEM #15: Featurerequest: Support East German robotron PC

mgoegel, 2024-06-20:
Hi Freddy,

this is not a real issue...
I started recently some projects for a robotron EC1834 8086 computer, which has a different bus connector.
In the Forum robotrontechnik.de someone announced your project and there was the question, if we could adopt it for these computers too.
I have no other idea, how to get in contact with you, so I choose this way.

If you want - we can discuss things via mail (m.goegel@gmail.com).

Regards Mario

FreddyVRetro, 2024-06-27:
Hi,
It seems to be an extremely rare machine. 9%  Chance I will not so a picomem for it.
Is there a way to do a converter between this bus and ISA ?

mgoegel, 2024-06-28:
Hi Freddy, thanks for your response.
I know, it's a rare machine.

Yes, I have build an 8 and 16 bit ISA Adapter for it, the problem is, that, because of the form factor, you can't close the case with it. Your card looks high, probably too high, to add the adapter and want to close the case.

I started 2 years ago with restoration of 3 of these computers. If you look in my repo, you can see, that I sucessfully adopted a XTIDE and a NE2000 board (and other things) for that computer. So, I have at least some experience with creating these boards.

The only thing I need for that is schematics... ;)

I'm pretty sure, your Pico code would run on the computer as on any other.

Regards Mario

Piligrim-77, 2024-06-28:
I join the question, somewhere can see a schematic diagram of picomem. Or is this information unavailable?

FreddyVRetro, 2024-06-28:
Hi,

I am working on the PicoMEM Low profile version.
I received the prototypes and may test it this week end.

Regards,
FreddyV

FreddyVRetro, 2024-07-11:
> I join the question, somewhere can see a schematic diagram of picomem. Or is this information unavailable?

Hi,
It will be published for sure, but I don't know when as well.
I mean when I will be ready to do so.

FreddyVRetro, 2024-07-11:
> Hi Freddy,
> 
> this is not a real issue... I started recently some projects for a robotron EC1834 8086 computer, which has a different bus connector. In the Forum robotrontechnik.de someone announced your project and there was the question, if we could adopt it for these computers too. I have no other idea, how to get in contact with you, so I choose this way.
> 
> If you want - we can discuss things via mail ([m.goegel@gmail.com](mailto:m.goegel@gmail.com)).
> 
> Regards Mario

Do you think that the PicoMEM Low profile will be sufficient for your need ?
I will surely not do picomem for really rare and not ISA compatible PC.
Except Tandy 1000 EX/HX

mgoegel, 2024-07-11:
I don't know - probably. What is the form factor?