FreddyVRetro/ISA-PicoMEM #17: Monotech NuXT picomem not recognized

jeroenvanlieshout, 2024-06-27:
I have a NuXT (https://github.com/monotech/NuXTv2/tree/master/Utils/BIOS), a bit comparable with the Book8088, but then fully open source with the 1.0 BIOS of Sergey Kiselev and the R625 XT-IDE BIOS. It seems that somehow the PicoMem is not recognized at all. I tried to Book8088 sample config.txt as I think the architecture looks a bit like that. Is this a known incompatibility as it is not on the list yet?
Also good to mention is that I clocked down my V20 cpu to 4.77 MHz.

FreddyVRetro, 2024-06-27:
Hi,

Does the PicoMEM Power on ? (LED ON)

jeroenvanlieshout, 2024-06-27:
Yes it does (green light), but that takes a while before that happens. The normal boot process is then already booted into dos. Good to mention is that I flashed the latest firmware and have a brand new board ordered from texelec.

jeroenvanlieshout, 2024-06-29:
[image]

More details about the boot process of this particular bios config.

FreddyVRetro, 2024-07-22:
Hi,

You can test the PicoMEM Fast Ram firmware now, maybe it will solve the problem. (But it remove EMS)

jeroenvanlieshout, 2024-07-30:
Unfortunately it didn't help here. The whole thing doesn't get recognized. Could it also be the case that the BIOS just doesn't support the range that the PicoMEM is located in?

FreddyVRetro, 2024-08-11:
Hi,
The bios is standard on these boards normally.

If you boot in another way, you can check if the picomem bios is present with debug
"D d000:0000" for example

jeroenvanlieshout, 2024-08-13:
Found the culprit after running this command. There is on the NuXT a switch block to enable upper memory. This maps over the PicoMem when set incorrectly. Maybe good to mention that SW2 switch 3 must be set to disabled on the motherboard to prevent overwriting the BIOS with memory. This is to disable memory in area D000-D800. See also: https://github.com/monotech/NuXT/blob/8de6d0b00ee6e4f7bb28740c9396e35a298eac63/Monotech%20NuXT%20User%20Manual.pdf

After configuring this switch correctly the PicoMem firmware loaded straight up and even works together with the built-in XT-IDE BIOS.

FreddyVRetro, 2024-08-13:
Cool, I added a section, for the NuXT in the Wiki

closed 2024-08-13 by FreddyVRetro