FreddyVRetro/ISA-PicoMEM #19: IBM Model 30 8086 - PS2 Mouse not working

jdredd87, 2024-06-27:
PicoMem v1.11
May 28 firmware

IBM Model 30 - 8086

With the card installed, the onboard PS2 mouse will not work.

I take it the Mouse PS/2 port is shared with IRQ1 ? 

https://www.vogons.org/viewtopic.php?t=76815

If I unplug card, mouse works again.

CH375 USB card and VGA card are installed with the PicoMem.

jdredd87, 2024-06-29:
Some things i've done today for testing

1)  Remove PicoMem card - Booted off MS-DOS 6.22 boot disk.  CTMouse.exe - Mouse works.

2) Change IRQ jumper on PicoMem board 3,5,7.  No changes. Mouse does not work.

3) EMS - Disabled
    WIFI - Disabled
    ADLIB - Disabled
    Mouse does not work still.

Maybe more info on this 2nd post I found @ https://forum.vcfed.org/index.php?threads/vintage-mice-for-the-first-ibm-computers-and-clones.35106/#post-429548

rmpel, 2024-09-03:
I have a similar issue which will document, but just for completeness of information; my computer crashes when I mode my mouse. Does not matter if I use Microsoft mouse driver, Logitech driver or CuteMouse. As soon as I move the mouse, the pc is dead.
Equally; with all functions of the picomem disabled.

rmpel, 2025-12-16:
I'm very sorry, I thought I already commented on this. It's been a while, but I remember the mouse working. I then continued to add virtual harddisks, and during that, my computer broke, power supply "exploded", so I turned it off and shelved it for future repairs and haven't looked at it since. So is the problem solved? yes, I believe it is, thank you ever so much! Can I be 100% certain? no, unfortunately, not as I have no way to test it at this time.

FreddyVRetro, 2026-10-12:
Old Post,
Feel free to open a new one if needed

closed 2026-10-12 by FreddyVRetro