FreddyVRetro/ISA-PicoMEM #21: EMS and USB Host settings get lost after power off.

jdredd87, 2024-06-27:
PicoMem v1.11
May 28 firmware

IBM Model 30 - 8086

EMS and USB Host settings are reset after powering off.

FreddyVRetro, 2024-06-28:
EMS settings lost is corrected (Next firmware)

Gr8Britton, 2024-06-29:
> EMS settings lost is corrected (Next firmware)

Thank you, @FreddyVRetro !  I just got my PicoMem v1.11 in the mail a few days ago and was thinking is was just me!  :-)  Do you have date you are expecting to release the next firmware?

Thanks for the great work on this card.  You have brought my deceased father's Sinclair PC500SD back from dead since the 5.25" drive didn't work!

FreddyVRetro, 2024-07-11:
Hi,

Next firmware I hope before end of next week.
I close this issue as corrected in next firmware

pawosm-arm, 2024-07-30:
I'm still observing USB host setting not being remembered. I'm setting it to 'off', and it is back on after power cycle.