FreddyVRetro/ISA-PicoMEM #23: EMS - Not sure if bug - Found / Not found depending on settings / software to init

jdredd87, 2024-06-27:
PicoMem v1.11
May 28 firmware

IBM Model 30 - 8086

EMS Port  = 268
EMS Addr = E2000

PMEMM.exe - Will not find the memory at all.
LTEMM.exe - will find it. But only 2mb established.  ( I have a lotech 2mb card i removed, and this was the settings for it also )

Change it up

EMS Port  = 2A8
EMS Addr = E2000

PMEMM.exe - Auto finds and all 4mb shows up.
LTEMM.exe - did not test after PMEMM.exe worked - I can retry if desired. To see what it does?

So problem comes down to maybe, why did PMEMM fail but LTEMM worked with at port 268 vs 2A8?

FreddyVRetro, 2024-06-27:
Hi,

LTEMM Does not check the EMS Port, so it say detected but in reality it is wrong.

The Port 268 seems to be used on your PC, so everything is normal there

FreddyVRetro, 2024-06-27:
Not a bug, PMEMM detect that th eport is in conflict, not LTEMM

closed 2024-06-27 by FreddyVRetro

jdredd87, 2024-06-27:
OK wasnt sure.  As the LTEMM "worked" at first at 2mb when i defined its setting in config.sys and the PMEMM.exe wouldn't work.