FreddyVRetro/ISA-PicoMEM #24: Hard Drive Corrupted

jdredd87, 2024-06-27:
PicoMem v1.11
May 28 firmware

IBM Model 30 - 8086

To be honest, not 100% sure what I did. And have not gone back to repeat it yet.

Put the card in, went to PicoMem BIOS and checked it out.

Booted system then, and the MS-DOS 6.22 boot up process dumped all kinds of random data to screen.

After reboot, no more booting.  

Removed PicoMem - No boot still.

Nothing important lost but figured mention this.  

I can restage things up, but looking for anything you want me to do first to stage it all up?

FreddyVRetro, 2024-07-11:
Hi,
Does it really corrupt the drive or only show corrupted data when doing DIR ?

FreddyVRetro, 2024-07-22:
A conflict between I/O and DMA was found and corrected, maybe it solve this issue. 

I close the case.

closed 2024-07-22 by FreddyVRetro