FreddyVRetro/ISA-PicoMEM #25: USB - Mouse - Does not go up or down.

jdredd87, 2024-07-02:
PicoMem v1.11
May 28 firmware

IBM Model 30 - 8086

Got A USB OTG adapter for A USB mouse.

Using PMMOUSE it "works" as in it detects the mouse. 

I can move mouse cursor left and right. It detects left / right / middle wheel clicks... but mouse cursor won't go up or down.

Tried a few mice and same issue.

Not 100% sure but this happens with USB Host set to ON and OFF.  

I thought if OFF, it wouldn't work at all.. but it did... nothing changed.

I have the other log for the PS2 mouse not working also.

I have tested A serial mouse and it does work fine.

jdredd87, 2024-07-09:
It is odd... since the USB OTG don't work... and for whatever reasons the onboard PS2 port for the mouse won't work right either with the PicoMem installed.... power draw issue? something the PicoMem is hammering on that the IBM Model 30 don't like?  Anything about PicoMem and IRQ1 ?  

Using serial mouse isn't end of the world.... but do miss using an optical mouse on it...

FreddyVRetro, 2024-08-11:
Hi 

Sorry for the late answer I am on holliday.
Does everything else work ?

Did you try to use another interrupt (on board jumper)

FreddyVRetro, 2024-10-11:
Hi @jdredd87 
did you test with the latest firmware ?

antoniocmateus, 2024-11-21:
Definitely something going on. I bought a ps/2-usb active adapter, which have a usb on one end and 2 ps/2 connectors on the other side. mouse only goes up and down, not left and right. Buttons work.

If you need more info please let me know

DevilsJoker, 2024-12-27:
When I use a USB mouse i can only move it up and down, but if I use a USB to PS/2 converter and a PS/2 mouse it works fine.