FreddyVRetro/ISA-PicoMEM #26: Sinclair PC500SD / Amstrad PC1512 Will Only Boot From Floppy

Gr8Britton, 2024-07-04:
I have your card installed and working.  (LOVE IT!)  My only issue is that if I have a FDD mounted, it boots from floppy (as expected.)  However, if I do not mount a floppy but do mount a HDD image, it then wants to boot form the physical floppy drive (which is not working.)  Am I doing something wrong?  Or, is this just a nature of the beast with this old PC?
Thanks!

FreddyVRetro, 2024-07-08:
Hi,

Maybe your disk image is not bootable ?
Does it display the image name at the startup ?

The PicoMEM is developped on a PC200 and I test it regularly with PC1512 / PC1640, there is no problem on them.

Gr8Britton, 2024-07-08:
Thank you for the response.  As it turns out, it was an **ID10T** error!  Somehow I had turned off the PicoMem boot option.  I turned that on and it suddenly worked properly.  Imagine that?!?!  Sorry and thank you!

FreddyVRetro, 2024-07-11:
Working.