FreddyVRetro/ISA-PicoMEM #27: BIOS Setup Tool not usable on mono-chrome devices

a3nw3n, 2024-07-06:
I try to use the picomem Card on an ancient Olivetti M24, which in general seems to work fine.

But as this 8086 device only has a text / green monochrome display, the UI of the setup tool is barely usable (see Screenshots). The the brightness or contrast dials sadly habe no effect.

Is there some config option to use an b/w mode or other colors scheme with more contrast?

Just by guessing the keystrokes I managed to get it working, but changing disks is hardly possible.

Thanks!

[image]

[image]

[image]

jschulbach, 2024-07-08:
Amazing, I literally hit this same issue this weekend trying out the picomem on my PC 6300(rebadge of the Olivetti M24). I'm seeing the same missing text on my CRT 313/H monochrome display. Came here to report an issue and you beat me to it!

vkoskiv, 2024-07-08:
Encountered this exact same thing just now with my 5151 and a Video-7 VEGA card set to MDA mode

nikolarun, 2024-07-08:
Just had the same issue on a 8086 NEC V40 with a Trident 9000c with VGA on a LCD monitor. Would love to know if there is a config i can use

FreddyVRetro, 2024-07-08:
Hi,

In next firmware release I will compile one with the "Light blue on Grey" color changed (it is the one not visible)

nikolarun, 2024-07-08:
> Just had the same issue on a 8086 NEC V40 with a Trident 9000c with VGA on a LCD monitor. Would love to know if there is a config i can use

My issue turned out to be completely unrelated. when taking the case off to install the card , i had changed the switch for the video output on the integrated CGI to Colour, once I switched it back to Mono it was fine. ( I had no idea that that switch impacted my ISA Vga card... )

FreddyVRetro, 2024-07-22:
Hi,

If you read this, I released a variant of the firmware for Monochrome screens.
Can you test it please ? :)

jschulbach, 2024-07-22:
Looks good on my PC 6300 so far. Will test further this evening.

[image]
[image]
[image]

JeffWDH, 2024-10-16:
@FreddyVRetro would it be possible to get a monochrome release of the Oct 5th firmware? Thanks!

FreddyVRetro, 2024-11-14:
> @FreddyVRetro would it be possible to get a monochrome release of the Oct 5th firmware? Thanks!

@JeffWDH Yes of course, I will try to not forget
Later, I hope to do it selectable in the BIOS itself.

closed 2024-11-14 by FreddyVRetro

Pernotto, 2026-10-10:
I just bought a picomem2 for my olivetti M24 and I have the same issue, but there is no "MD : Monochrome screen version" for its firmware yet.

Pernotto, 2026-10-10:
I connected my color monitor, and it appears that the problem is not limited to monochrome monitors.

[image]