FreddyVRetro/ISA-PicoMEM #28: Floppy exchange without PMINIT

SelfRef, 2024-07-18:
Hello,

I'm tinkering with PicoMEM for some time and I noticed the floppy emulation functionality could be slightly improved by providing non-software way of swapping disks. I'm aware of using PMINIT that's very convenient to select desired image on running system, but it has disadvantage of not playing well with multi-disks installations of DOS and (even more) with multi-disks software that boots directly from floppies.

For such purpose I always used Gotek. PicoMEM could supersede Gotek, but that would require some way to exchange floppies on the fly by hardware. That wouldn't even require selecting a specific image in most cases, because a single rotation between images would be good enough.

On Goteks various mods can be added, including an additional button. I already see Pico can be extended with DAC and HEX display for POST codes. Is it possible then to add a hardware button support to be connected to some pins and act as a floppy switcher? Hex display could also be repurposed for displaying active image index.

Additional thoughts:
I recently modded my Gotek to use [rotary switch](https://hxc2001.com/docs/gotek-floppy-emulator-hxc-firmware/pages/addon-buttons-rotary.html) and [OLED display](https://hxc2001.com/docs/gotek-floppy-emulator-hxc-firmware/pages/addon-oled.html) to use built-in menu and changing images. Both use just a single I2C interface and are easy to add. Maybe this is also the good idea for PicoMEM to apply dinamic changes to configuration without boot menu?

dotelpenguin, 2024-07-18:
I was thinking about this the other day while trying to install OS/2.   A single "NEXT"  button could be used.  
Name your disk images in a way that can be incremented.
OS2-1, OS2-2, OS-3, etc.  and on each button push it rotates to the next image. This way if you  have lots of images it will only  rotate through the set you defined by name.  Or maybe perhaps sub-directories.

FreddyVRetro, 2024-07-22:
Hi,

This is something I actually wanted to do before doing the actual floppy change code.

I2C enter in conflict with the Audio and code I2C button / screen take lot of time, I prefere to spend time on other fonctions for the moment.

You can also use PCEM to install OS and application on PicoMEM images, it is really simple.

SelfRef, 2024-07-22:
I'm glad you thought about this. Definitely that's not something important in comparison to other features, but definitely worth to place on the backlog. I already use 86Box very extensively in tandem with PicoMEM, and it indeed makes everything easier.

Furthermore, I talked with my friends about some features. We only use PicoMEM as a temporary (yet essential!) tool while repairing/preparing old PCs, and we agreed the [PicoGUS](https://github.com/polpo/picogus) is more appealing if someone is looking for sound abilities. So even if some features will collide with each other, they may be very handy as temporary debugging functions.