FreddyVRetro/ISA-PicoMEM #30: IBM 5150 - not seeing option to enter PicoMEM BIOS

JeffWDH, 2024-08-01:
Firmware: PM_W_Jul16_MD.uf2

I just received this PicoMEM and installed it into an IBM 5150. I'm not getting an option to enter the PicoMEM BIOS at boot and the system goes straight into BASIC. I've tried a line with "PREBOOT" to config.txt but that doesn't seem to do anything. 

Any advice?
Thanks!

JeffWDH, 2024-08-01:
Just as a sanity check, do you think I'm running up against this:

```
Symptom 	Does not work in an early IBM 5150 (IBM PC).
Background 	The XT-IDE card uses a 'BIOS expansion ROM' (the EEPROM on the card) (a.k.a. BIOS ROM) (a.k.a. boot ROM) (a.k.a. option ROM).
Cause 	        The motherboard BIOS supplied in an early IBM 5150 does not support BIOS expansion ROM's.
Fix            	Upgrade the motherboard BIOS to the final one (revision 10/27/82).  That BIOS supports BIOS expansion ROM's.
```
https://minuszerodegrees.net/xtide/XT-IDE%20-%20Problems.htm

FreddyVRetro, 2024-08-01:
Hi,
Picomem bios is also an expansion ROM, so I think yes.

JeffWDH, 2024-08-07:
Can confirm, the 10/27/82 BIOS is the fix.

FreddyVRetro, 2024-08-16:
Thanks for your help, entry added to the Wiki

closed 2024-08-16 by FreddyVRetro