FreddyVRetro/ISA-PicoMEM #32: PCI bus

alexartwww, 2024-08-04:
Why didn't you add PCI port from the other side of this card?

E.g. it could be useful if I want to install OS on powerful PC and then insert it to 286.

Is it possible?

megatog615, 2024-08-04:
Not what I expected from the title but it would be nice if the picomem had support for 16 or 32-bit buses like 16-bit ISA or 32-bit PCI or even VLB(but I'm not holding out for that one).

FreddyVRetro, 2024-08-11:
Can t be done

closed 2024-08-11 by FreddyVRetro

mcgurk, 2024-09-16:
Just install OS with powerful PC and read drive to imagefile to microSD-card with modern PC.

If more powerful PC includes IDE, use cheap CF-IDE-adapter and CF-card.
If more powerful PC includes SATA, use cheap USB-SATA-adapter.

Even easier is to install OS with modern PC and modern OS with virtual machine or emulator.