FreddyVRetro/ISA-PicoMEM #35: CTRL-ALT-DEL No longer works properly

jkmills78, 2024-08-08:
Firmware: PM_W_Jul16_FR
Hardware: v1.11
Computer: Hyundai 386Super-STC 386 20MHz
More details upon request.

When the computer boots from the PicoMem, CTRL-ALT-DEL causes the compuyer to issue 3 slow beeps followed by the keyboard no longer responding.  The computer doesn't appear to be locked up as the cursor still blinks.  When booted from a floppy drive, with everything else the same as before (PicoMem installed, sd card in, etc) CTRL-ALT-DEL then works as normal.  However I noticed that I could no longer access the PicoMem C: drive when booted from a floppy either.

FreddyVRetro, 2024-08-11:
Hi,

If it fail to rom error,  Maybe you enabled rom mirroring ?
Can you try another nios base address ?

jkmills78, 2024-08-15:
So far I have tested three addresses: C800, D000, and D800.  No change in the behavior.  I have confirmed that in my BIOS settings the shadow BIOS ROM is turned off (already set this way, so no change in settings).  Are there any other settings that you would like for me to test?

FreddyVRetro, 2024-08-16:
Did you test the Fast RAM firmware ?
Is there BIOS settings where you can add Wait states to the ISA bus ?

jkmills78, 2024-08-17:
The firmware I used is the latest Fast RAM firmware.  In order to provide you with additional information about my environment (and also hopefully answering your question), I am including pictures of all of my current settings for both the PicoMem and my system's BIOS.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jkmills78, 2024-10-30:
I had tried all port values before with no change, including C800.  EMS is disabled in the BIOS, so I don't think that makes any difference what port it is on.  This motherboard does not support ISA timing changes I am guessing because I don't see anything in the BIOS.  I posted all of my BIOS settings screens so what you see here is what you get unfortunately.  I would really like to get this to work, so I will try to change the EMS port and I will also try changing the ports in the picomem again just to double check myself.  Maybe I missed something the first time around.  I will let you know the results.

jkmills78, 2024-11-03:
As it turns out, this isn't a problem with the picomem at all.  After going through hours of many changes to both my BIOS and the picomem config itself, along with deleting the PICOMEM.CFG file each time, and trying various combinations of things such as different images, I was no closer to getting it to work. I finally decided to comment out everything in my autoexec.bat and config.sys files and in the end it turns out to be some kind of issue with EMM386.  There is a simple parameter `ALTBOOT` that can be passed to the EMM386 executable which eliminates this issue.  Here is a link to the issue if you are curious: [https://jeffpar.github.io/kbarchive/kb/093/Q93366/](https://jeffpar.github.io/kbarchive/kb/093/Q93366/).  This issue can be closed, and I apologize for wasting your time.

closed 2025-12-16 by FreddyVRetro