FreddyVRetro/ISA-PicoMEM #36: A2386 - Not seeing option to enter PicoMEM BIOS

ChrisHammond56, 2024-08-08:
I purchased the PicoMEM based on the compatibility list (tested with A2286/A2386 in A2000). However, I am unable to enter the BIOS setup screen.

I have disabled Janus on the Amiga (D000 range disabled) just in case, and even tried the BIOS C800 in CONFIG.TXT, and PREBOOT option but none of the options seem to work. 

Can you enlighten me as to how the configuration is setup for an A2000 with A2386 BridgeBoard.

Thanks

FreddyVRetro, 2024-08-11:
Hi 

Do you see at least the PicoMEM initiation  message ?

I may need to find who got it working on A2286
I cant now, I am on holliday.

andabata, 2024-08-11:
I do have it working on an A2286, but i don't have an A2386 to test

ChrisHammond56, 2024-08-12:
I have been testing some more over the weekend. The good news is it is now working, but I have no idea how I managed to get it so. I did the following:

1. Disabled JANUS in Amiga (but BIOS C800 was used anyway)
2. Removed VGA Card
3. Removed ISA Serial/Parallel Card
4. Card was then detected.
5. Re-added VGA card - Still Detected
6. Enabled JANUS as D000 - Still Detected
7. Re-added ISA Serial/Parallel - Still Detected

So I am unsure what actually fixed it?

Thanks for the feedback anyway, and have a great holiday.

kp73, 2024-08-13:
Has anyone got this working with an A2088 XT 8088 bridgeboard?  I tried and have a similar problem, no option for 'BIOS' or PicoMEM initiation message.

FreddyVRetro, 2024-08-16:
Hi,

It had another report it did not work on A2088.
Maybe you can try the "Fast RAM2 Firmware version ?

kp73, 2024-08-16:
Will do

kp73, 2024-08-19:
[image]

 Thanks for the helpful suggestion and a great project. 

Some success.  It's now displaying the bios message but I can’t hit ‘S’ as Janus hasn’t loaded - so the Amiga keyboard isn’t working 🤦‍♂️

ChrisHammond56, 2024-08-19:
You don't need Janus to be loaded for the keyboard to work. Did you open a Monochrome or Colour window on the Amiga side before switching to the PC side? Are you using a VGA card? If so, then just having the monitor switched to the VGA output doesn't run anything on the Amiga to allow comms. I use a VGA card and in the BB prefs, I have Janus loaded at D000, Monochrome Enabled and Colour Disabled. Then on the Amiga side, run the PC Monochrome, which will open a window on the Amiga. Ensure that window is active/focused, and then you can then switch to the VGA output on the BB and input will work for the keyboard.

kp73, 2024-08-19:
Yeah, I have the same setup.  VGA with mono window opened and focused on the Amiga side.  The keyboard doesn’t respond- or at least the PicoMem BIOS doesn’t open when I hit ‘S’.  After Janus I s loaded I get a load of beeps from the keyboard after I hit ‘S’ several times.

kp73, 2024-08-19:
It’s working!  Thanks all for the help.

So all I did was stick the PicoMem in another machine,
[image]
 changed the ‘PicoMem Boot Code: on’ and it worked when I put it back in the A2000.

Entering the PicoMem BIOS seems to work fine too now, not sure what was up with it before.  These bridge board seem a tad flaky :) but it’s working now so I’m happy.

FreddyVRetro, 2024-10-04:
Hi,

Nice.

Change this settings should not make any difference in the fact that the keyboard did not work.

It is even more difficult for me to understand that I don't know how the PC boards work on Amiga.

One friend told me that JLINKS use the space between 512K and 640Kb to do the link with the Amiga, so you should not add RAM in this zone to use it.

rkbrooks, 2024-12-09:
I can say that I made a board a few years ago to add the 512k -> 640k ram on a A2088 and did not run in to problems with any aspect of Janus,