FreddyVRetro/ISA-PicoMEM #37: IBM 5150 will not boot from emulated hard drive

ifilot, 2024-08-09:
My IBM 5150 is unable to boot from the emulated hard drive.

1. I was able to emulate a hard drive (20Mb), partition and format it and install MSDOS (5.0) onto it. 
2. When booting the computer, the floppy drive is being probed whether it has a bootable medium. When this fails, instead of probing the hard drive, the ROM BASIC is launched.
3. When a bootable medium is present and the computer boots from the floppy drive, I am able to access the emulated hard drive and see all the files installed, which work without problem.

Relevant details to mention:
I am using a Gotek to emulate a floppy drive. This Gotek is hooked up to the original floppy drive controller.

I appreciate any suggestions this community has to offer to resolve this issue. :-)

FreddyVRetro, 2024-08-09:
Hi,
Have you done fdisk /mbr to install the partition loader ?

ifilot, 2024-08-10:
Thanks for the quick reply!

Initially, I did not do this. I was under the impression that the MS-DOS installation would take care of this. I did so afterwards, i.e. booting the system via the floppy drive and then running `fdisk /mbr`. It however does not solve the problem.

FreddyVRetro, 2024-08-11:
Hi,

Did you try the images I provided on github ?
How did you create the 20mb image ?

ifilot, 2024-08-11:
* I did so far only try the 6.22 image which did also not boot. I have not yet tried the other ones.
* I created the 20mb image via the option in the PicoMem BIOS, attached it as drive 0, then run `fdisk` to partition it and finally format it. Next, I booted from the MS-DOS 5.0 installer floppy using the Gotek and installed MS-DOS on the `C:` drive. The reason I use MS-DOS 5.0 is because it is known to work using a physical hard drive that I own as well.

Basically, everything is working nicely, except the booting from the emulated hard drive that it. I will try the other images (will be next week though due to other obligations), but I'll let you know.

Thanks once more for assisting in the troubleshooting. Appreciated!

FreddyVRetro, 2024-08-11:
Hi,
So you probably did fisable picomem boot code.
You should enable it.

ifilot, 2024-08-17:
Correct, that was the mistake I made. Turning this on resolves the issue.

Thank you so much for the kind help!

closed 2024-08-17 by ifilot