FreddyVRetro/ISA-PicoMEM #38: PCM5102 Adlib Instructions

SanPollo, 2024-08-12:
I have bought a PCM5102 board to connect to my PicoMEM for Adlib emulation.

However, I saw on a [ctrl-alt-rees video](https://youtu.be/RACp9P-KUVE?si=MG2EdrXGC90rZ4XP&t=36) that the board must be modified with a soldering iron to work with the PicoMEM. Unfortunately, I can't find any instructions in the repo, or on the wiki.

Please would you provide instructions for modifying/configuring the PCM5102 to work with the PicoMEM, and also update the documentation on the wiki?

xinteksik, 2024-08-13:
[image]

[image]

FreddyVRetro, 2024-08-16:
I added an audio section in the Wiki

closed 2024-08-16 by FreddyVRetro

SanPollo, 2024-08-16:
Thanks both!