FreddyVRetro/ISA-PicoMEM #39: Southamerica Distribution of ISA-PicoMEM

nsinav, 2024-08-13:
Hello:

Hi, I'm sorry to write to you through this medium, but it's the only way I found. I'm interested in distributing this product in South America. Is this possible? Thank you in advance for your reply.

FreddyVRetro, 2024-08-13:
Hi,
I promiced exclusive distribution to Texelec for America, sorry :-(

nsinav, 2024-08-15:
Hello. I understand. And maybe distribution only in Chile? (My country). I think Kevin Williams does not care about South América, maybe i can talk with him, i got the distribution for PETSCII Robots and Planet X3 from his partner David Murray. But I prefer to keep the local market, as it is a good market for retro hardware. I am really interested. Kind regards,

iosoft, 2024-09-01:
why not sell directly through eBay International?

closed 2024-12-03 by FreddyVRetro