FreddyVRetro/ISA-PicoMEM #41: Commodore COLT (PC-III / PC-10): Pi Pico does not boot without XT-IDE

jtgans, 2024-08-18:
So, this is a weird one. I have a Commodore COLT (just a PC-10 motherboard) and the Picomem installed in the slot furthest from the CPU, running the latest firmware with an SD card present. If I *only* have this configuration and power on, the Pi Pico will not boot (no green LED). If, however, I install an XT-IDE next to the Picomem, suddenly the Pi Pico boots (I get a green LED).

If I plug the Pi Pico into my laptop via USB, it happily boots (green LED).

Any idea what might cause this to occur?

jtgans, 2024-08-18:
I have other option cards as well, like a Gamecard III Plus from CH Products. Doesn't seem to cause this effect. Also, the positioning of the cards doesn't seem to matter, either.

FreddyVRetro, 2024-08-18:
Hi,

The Pi Pico is sensible to the power quality.
It is clearly a problem with the 5V that is not clean enaugh for the PicoMEM.

When you add the XTIDE board, it add some more filtering capacitors to the 5V that help the PicoMEM to start.

jtgans, 2024-08-19:
Sounds like you need more bulk capacitance on the PicoMEM, if that's the case.

jtgans, 2024-08-21:
Yeah, reviewing the board, you have decoupling caps for every chip, but none for the Pi Pico itself aside from the SMPS you have in place below it, and that isn't nearly enough. You want at least one bulk cap onboard, 10nF at least when using a Pico board.

I've verified the rails on my PSU, 5v and 12v are both absolutely rock solid, with nearly no measurable ripple or sag. Haven't verified 5v at the backplane with just the pico yet. If I get a chance I'll see if I can get a trace of the system's 5v contrasted with your SMPS 3v3 power rail coming up at start of day with just the Pico installed.

FreddyVRetro, 2024-08-26:
Hi,

Yet it is stable on my PC10 and I know tyo other person that tested it on PC10 /PC20 Ok

Did you try to add a capacitor on the PicoMEM to test ?

FreddyVRetro, 2024-12-03:
Hi,
Do you have any news or progress ?

Cloudscout, 2025-05-06:
I'm seeing the same issue with my PC10-III.  The PicoMEM will only power up if I connect external USB power cable.  I'd be happy to try adding a capacitor.  What value should I try?

FreddyVRetro, 2025-05-06:
> I'm seeing the same issue with my PC10-III. The PicoMEM will only power up if I connect external USB power cable. I'd be happy to try adding a capacitor. What value should I try?

Hi,

I added U10 so that lot of PC are able to Boot, this chip "Reproduce" a Power good signal for the Pico PSU to start when the 5V is stable.
This chip can add problem on other machines, if the 5V is too low.
It was extremely tricky for me to test, I tried, 5x different chip.

So, Try to remove U10 (But tricky to put back if needed), add a "Big" capacitor, 100, 200nf or see if the PC PSU need a recap.