FreddyVRetro/ISA-PicoMEM #42: PicoMem BIOS not detected after hot reboot

jeroenvanlieshout, 2024-08-23:
Now I got it working on my NuXT I see behavior that after hot reboot using ctrl+alt+del it never detects PicoMem. Only way to boot it again with PicoMem is doing a cold restart. Using the FastRAM version as there is a V20 CPU with higher clockspeed setting installed. However also on slower clockspeeds the behavior is the same. 

Good to mention: XT-IDE is located at F000 and using PREBOOT option it works with PicoMem at D000. Could it be that the behavior or timing can prevent that it is being recognized at hot reboot?

FreddyVRetro, 2024-08-26:
Hi,

PicoMEM not recognized after a Hot reboot may be due to something crashing it.

If you are able to boot onthe NuXT anyway, with an XTIDE, you can check if the PicoMEM is alive with multiple IOR to the port 2A3h
then, display the PicoMEM BIOS region:
debug
i 2a3 (multiple time)
i 2a0 should answer 0 (It means ready)
then:
d d000:000
to see if the PicoMEM BIOS is visible

It can be code in the BIOS using the port 2A0 for example (PicoMEM command)

jeroenvanlieshout, 2024-08-26:
I get both when booting hot or cold same results with debug commands. I 2a0 returns in both cases 00. Also I can see with the debug command the BIOS in the memory space. Only for some reason it looks like at a hot reboot it is somehow not recognized by the BIOS.

FreddyVRetro, 2024-09-01:
Maybe someone with BIOS Knowledge on the NuXT can help debugging it.
Do you have contact with its BIOS Creator ?

luc-dc, 2024-12-10:
I had the same problem on the micro8088 and got it working by modifying the bios to disable the checksum verification.
After this, the picomem bios is always shown.
Strangely, when I dump the bios after a cold boot and after a reboot and compare the two, there is no difference. Only in the memory after the rom. Wonder if the checksum routine takes that into account while it shouldn't.
I'm using the Oct5_FR firmware, the normal Oct05 doesn't work.