FreddyVRetro/ISA-PicoMEM #45: Can't open wifi

GeCro1950, 2024-08-28:
No matter what I try, I cannot activate the WiFi. 
I keep getting the message Can't open wifi.txt.
My text file is named: wifi.txt
What am I doing wrong?

Station2381-sys, 2024-08-29:
the file must be in the root of the card.  the first line is the SSID name and the second line is the password.  so check if the wifi.txt file is there.  if you created it in Windows, it can be called wifi.txt.txt.

GeCro1950, 2024-08-29:
[image]
[image]

Station2381-sys, 2024-08-29:
delete the .txt and then you'll be fine.

Station2381-sys, 2024-08-29:
[image]

GeCro1950, 2024-08-29:
Thanks Station2183-sys that was the problem....

closed 2024-08-29 by GeCro1950