FreddyVRetro/ISA-PicoMEM #47: Anyone know how to make a USB mouse work in Windows 3.0 with PicoMem

TimRustige, 2024-08-29:
Anyone know how to make a USB mouse work in Windows 3.0 with PicoMem

I've downloaded the mouse driver from here and in Dos it tests fine, all working.

Can't find any combination of default mouse driver options in Windows 3.0 that will give a working mouse though
with the PicoMem. Anyone else been here?

Station2381-sys, 2024-08-29:
I have the same problem.  the pmmouse driver allows different settings, you can try different combinations.

FreddyVRetro, 2024-09-01:
Hi,
I was aware of this, I think it is because the PicoMEM need "2" IRQ for it, and Windows probably disable the PicoMEM Hardware IRQ.
Maybe someone will be able to debug this in the future but not me, I don't know how windows work :(

antoniocmateus, 2024-11-21:
+1 for mouse not working on windows 3.1. Any way this could be addressed in future updates?

mdovey, 2024-12-03:
I don't think the problem is IRQs, but that pmmouse (and cutemouse upon which it is based) is a Dos TSR (Terminate Stay Resident), and these don't work under Windows.

This is also why pminit shortcut keys don't work under Windows.

Either the picomem mouse code from pmmouse needs reimplementing as a Windows 3 driver, or the PS/2 mouse emulation needs to be done entirely by the picomem rather than in software. The latter is probably the "best" solution in that it would then work for other "operating systems" (OS/2, Win2, Win31, Win95 etc.) rather than requiring a bespoke driver for each, but may not be the easiest to implement.

samrg123, 2026-10-04:
> For Windows 3.0 _**running in real-mode**_, Microsoft's own Windows mouse driver (MOUSE.DRV) can be modified to force the Int 33h passthrough support normally reserved for use with OS/2, which works in tandem with the PMMOUSE DOS driver.
> 
> Hex edit MOUSE.DRV and search for and modify the following routine:
> 
> From:
> 
> B4 30 CD 21 3C 0A _**72 05**_
> 
> To:
> 
> B4 30 CD 21 3C 0A _**EB 00**_

FYI this also works to patch the windows 2.03 MOUSE.DRV present in the installation setup floppy (first disc), but you'll need to patch it prior to installation or reinstall windows again. You can also replace the windows 1.01 MOUSE.DRV (also in its setup floppy) with the patched one from windows 2.03 to get the picomem to work in OG windows as well 😎