FreddyVRetro/ISA-PicoMEM #48: Schematics?

thetechknight, 2024-08-30:
Where are the schematics for this particular card? 

I am looking to figure out how to adapt this stuff to proprietary XT hardware such as the Toshiba series laptops, but I dont see any schematics available to determine which signals are used and not used. 

Thanks.

chadr, 2024-10-14:
> Where are the schematics for this particular card?
> 
> I am looking to figure out how to adapt this stuff to proprietary XT hardware such as the Toshiba series laptops, but I dont see any schematics available to determine which signals are used and not used.
> 
> Thanks.

They haven't open sourced the schematics because it's a cash grab.

FreddyVRetro, 2024-10-14:
@chadr I worked on Mod Master XT during 10 years and did no money at all with it.
Source is published.
I worked 2 years on the PicoMEM, if you add the cost of all the boards it emulate it is more than 200€.
I have a wife and 3 kids and I spend most of the spare time on it.
If I count the nb of hours spent on it, I am not sure it is a cash grab like you say....

FreddyVRetro, 2024-10-15:
@thetechknight 
Hi,

I don't release the schematics by now of course to keep control on the produced boards mainly for a support reason.

I may release the schematics of the Current PicoMEM Version when I have Pico2 design.

If you are motivated to do the design for the Toshiba, you are welcome to the Pico discord server and discuss about it.

closed 2025-05-06 by FreddyVRetro