FreddyVRetro/ISA-PicoMEM #49: Feature Request: RTC

iosoft, 2024-09-01:
Dear Freddy,

In next upgrade, pls add **RTC** feature.
It will help our Retro DOS based PCs to get the correct date-time automatically.
Its so boring to enter date-time on each start-up.

There are multiple cheap RTC modules which you can integrate.
(Example: DS1302 RTC Real Time Clock Module)

Thanks and regards.

MacGyverr, 2024-09-28:
Without a battery I'm not sure how this is possible, in the meantime you can set the clock to current this way.

I currently use:
`SET MTCPCFG=C:\MTCP\MTCP.CFG
C:\DRIVERS\NE2000.com 0x60 3 0x320
C:
CD\MTCP
DHCP
SNTP -SET pool.ntp.org`

in my autoexec.bat to add the network support, pull a DHCP address, then get and set the time from the pool.ntp.org website.
It can be ran in a simple batch file after boot if needed if you don't want to modify autoexec.bat.

FreddyVRetro, 2024-10-03:
Hi,

RTC is a best to have, so it should arrive, but no planned date at all.

closed 2024-10-03 by FreddyVRetro