FreddyVRetro/ISA-PicoMEM #50: Amstrad PC7386sx80 - computer crash when moving PS/2 mouse, regardless of mouse driver or PicoMem settings.

rmpel, 2024-09-03:
First of all, yes, I know this project is targeted at 8086/8088 computers, however, I find the project very promising and had hoped I could at least use it for extra HDD storage and NE2000 WiFi ;)

Purpose of this ticket; document that I have a problem, realising that my use case is far from the target audience, just to let other people know they can expect this.

If, however, a solution is found, I would be very very grateful.

Thanks for this awesome project, I will keep it safe for when I finally get hold of the XT machine I have been searching for for a good decade now.

**Tl;dr:**

Even with all functions of the PicoMem disabled, with all 3 hardware IRQs (3, 5, 7), with a PS/2 mouse installed, the computer will not work properly.
Amstrad pc7386sx80, 25 MHz, 387 coprocessor installed, 4 MB RAM, VGA on board, Vibra16 in ISA slot.
With QEMM a crash message is visible with option to reboot, with EMM386 or without EMM manager, as soon as the mouse is moved (PS/2 mouse with either Microsoft, Logitech or CuteMouse driver), the machine crashes.
It makes no difference if network, sound, storage or USB mouse on the PicoMem is enabled or not, or on which address/IRQ they are set.

More information for anyone interested ;)

The situation;

Amstrad PC7386sx80, a very small footprint machine (only 32 x 25 cm), with only 2 16 bit slots available.
I will not replace it with another machine as I have had this computer since I was 12 (1989) if I recall correctly.

Recently I replaced the spinning rust drive with a CF card (challenge as the BIOS only supports fixed types, now running Anydrive, working great), but after testing and trying multiple combinations of identical and/or different CF Card to IDE adapters, I canNOT get a second CF card, so expanding beyond 500 MB is difficult.

The PicoMem card would solve that problem.

One could say; use a second Multi I/O card; tried that, doesn't work, the BIOS does not support a second MIO card. I could try one with a BIOS IC (XT-IDE), however, I don't have such a card and I know there is a "ISA ROM Card" project, but then I am an ISA slot short.

I also would like to use my NAS as a network storage for easy transfer of software between old and new, and, well, I could use a real NE2000 card, but I have only 1 spare slot (SoundBlaster Vibra16 in the other, soon to be replaced with something even better ;) )

Finally, I had hoped I could use the 4MB the PicoMem provides in conjunction with the 4MB on the mainboard to get 8MB and maybe allow me to run Duke Nukem. (However, I was not expecting this would work or even if Duke would run on a 25 MHz 386SX)

So.

The PicoMem would solve a few issues; Network connectivity, Storage and maybe possibly memory expansion, and maybe even with some tinkering virtual CDRom...

Well, hopes, dreams, and in the end disappointment.

p.s. I'm still in the market for a Schneider Euro PC, for which this card would be ideal as that machine only has 1 8bit slot. I had two of those machines in the late 90s and, ..., yes ... we threw them away ... If only we knew ...

FreddyVRetro, 2024-09-05:
Hi,

There are options you can add in a config.txt file :

"BIOS 0" will have the BIOS emulation removed.
If you want to also remove the I/O port (2A0 to 2A3) you can add "OFF"

Then, by doing one and the other, it will confirm if the problem is the PicoMEM BIOS or I/O Port. (Conflict)

rmpel, 2024-09-05:
Thank you! I did see the 'configuration files' section in the wiki, but did not expect BIOS 0 would help, as, all functions work with the BIOS enabled, except for mouse, but without mouse driver, the computer works fine.
I will try it tonight when I am home from work.

Regarding your second hint, could you be a little more specific? do you mean `BIOS 0 OFF` in the config.txt?
If this is what you meant, then I will try that as well.

As far as I know, ps/2 devices live in the 0x60 and 0x64 memory space and IRQ 12, so I would not have suspected a conflict with the PicoMem, I will test tonight and let you know.

If you have more hints before then, I'll take those in consideration as well :)

rmpel, 2024-09-05:
With `BIOS 0` There are no problems, but, of course, also no functionality.
I then tried BIOS D000 and that did not help at all (so I guess not a memory conflict???)
Then I changed a setting, don't remember, sorry, it happend fast, it was the setting that said "Set OFF for XTIDE", I changed it to not be OFF, it gives a "B for Basic" or something like that, and now the PC reports a Fixed DIsk Controller failure if I have the card in.

Will need to reset the config file to stock, of course I did not make a backup... because, why should you ;)

rmpel, 2024-09-05:
I really have no clue what is happening now ...
I set `BIOS D000` as that should be the default address, on which it worked
I no longer see the BIOS banner of the PicoMem, unable to go to the setup
Computer reports a fixed disk controller failure. But since I can no longer access the settings, I cannot unselect the disk image
So I then removed the settings file (the PICOMEM file), doesn't help
I disabled the on-board IDE, suspecting a conflict, doesn't help
I then proceeded to reflash the firmware, hoping it would clear settings, but also, doesn't help
in all these tests, the IRQ jumper was set to 3, I also tried 7, but avoided 5 as that's where my Vibra16 lives.
To test, I removed the Vibra and set the jumper to 5, I then got a wirking PicoMem again, (though still the mouse issues)

So...

The problems I seem to have are now;
- IRQ conflict on IRQ 3 and 7, and no other options, other than 5 where my soundcard lives
- When using IRQ 3, very rarely the card boots, setup reports PicoMem IRQ to be 0, the very next boot makes the card unusable
- When using IRQ 7, "fixed disk controller failure", no bios banner
- When using IRQ 5, everything works for a short while, then audio corruption from the Vibra, the next boot, same problems
- In every situation Mouse movement causes a crash, though there should be no conflict as ps/2 are on 0x60 and 0x64 on IRQ 12

I really appreciate this project, but I think my computer just isn't suitable for it. 

If you want more information for debugging purposes or something like that, feel free to ask, but I consider this matter closed as "unfixable".

FreddyVRetro, 2024-09-05:
Hi,

To go back to initial state, Delete the config.txt and Picomem.cfg files.

I did not realized it crashed only with PS/2

Now that I know it crash on your amstrrad and an IBM, Maybe I will be able to reproduce the issue.
(By 1 or 2 months)

rmpel, 2024-09-06:
> To go back to initial state, Delete the config.txt and Picomem.cfg files.

-> Yep, that will do it (but somehow in my situation of many failings, it did not until -- most likely -- the IRQ=0 problem was solved, by removing the soundcard and setting the IRQ jumper to 5, see above.

> Also, did you try the "Fast RAM" mode firmware ?

-> I did, and I found a few bugs with it, but I think they are caused by all the errors and conflicts, so I don't think too much of them. (A number of fields, like the ram toggles) can be toggled from off to on, but you don't see "on", you do see all the P's in the memory map, but you cannot turn it off again, because the toggle still says "off". Toggling the second memory option on and off solves it. Stuff like that is happening. But again, probably not something that is reproducable.)

> I did not realized it crashed only with PS/2

Indeed, and only with the mouse. The PS/2 keyboard is working just fine.

FreddyVRetro, 2024-09-08:
One PicoMEM user got the PicoMEM working by removing / plugging back his Graphic card. (On an Amiga 2000 386 bridge board)
Some weired conflict may occur

rmpel, 2024-09-09:
Unfortunately, that's not an option for me as the VGA is integrated on the mainboard.

FreddyVRetro, 2024-10-03:
Pinned to try to not forget it :)

rmpel, 2024-10-03:
Oh, about almost forgetting. I recently got myself a 486, and it displays the same problem.

FreddyVRetro, 2024-10-04:
Same problem means PS2 crash ?

rmpel, 2024-10-04:
Yes; the card appears to work, but when the mouse is used (moved for example), the computer crashes.

FreddyVRetro, 2025-02-22:
> Yes; the card appears to work, but when the mouse is used (moved for example), the computer crashes.

Hi,
It is me again after a long time.

Got more report about the PS/2 problem.
I need to get my hands on a PC With PS/2 to test the PicoMEM with.

Did you try by removing the interrupt jumper ?

rmpel, 2025-03-15:
Hi @FreddyVRetro 
Sorry it took a while, my life has been stressful, and therefore tiring.

Today, I tested in my 386 with no IRQ jumper; the computer will then beep more on powerup (3 beeps instead of 1) and the PicoMem banner is not shown, hotkey does not work. Good thing; no crash when moving the mouse, but the card doesn't work.

On de 486, the card will initialize, i can use the setup, but the computer locks up at "Starting MS-Dos". 

So, no luck, I guess.

FreddyVRetro, 2025-05-13:
> Hi [@FreddyVRetro](https://github.com/FreddyVRetro) Sorry it took a while, my life has been stressful, and therefore tiring.
>
> Today, I tested in my 386 with no IRQ jumper; the computer will then [...]

Hi,

I found and correct the PS/2 problem now. Will be in next firmware release.

rmpel, 2025-05-15:
that's great news! looking forward to it!

rmpel, 2025-05-28:
fantastic, I'll be checking it out soon! Thanks for all the work and the notification :)

rmpel, 2025-05-31:
Can confirm, no more crash when moving the mouse. Virtual harddisk works, Now I need to re-learn how to NE2000 in Dos ;)