FreddyVRetro/ISA-PicoMEM #53: External ROM checksum incorrect - Amstrad PPC640

StolleMan, 2024-09-08:
Picomem won't work in my Amstrad PPC640. Other people seem to have success with this machine, I wonder if there is a problem with my hardware or my specific bios version?

I get an External ROM checksum incorrect on boot (states the address of the picomem).

XT-IDE bios works as expected

Any ideas on what to try?

FreddyVRetro, 2024-09-08:
Hi,

I has 2 other person who failed with a PPC640 and a PPC512, but 2 where it is working.
I received recently a PPC to ISA board, so I may check this in some month.

Did you test the Fast RAM Firmware and other ROM Address ?
Also, another Power supply ?

StolleMan, 2024-09-09:
Hi Freddy,

I've tried multiple addresses, I've tried the Fast RAM firmware and I have tried 2x different power supplies. The card works in my 5150. It feels like it's some kind of Amstrad foible...

StolleMan, 2024-09-21:
Update time:

I built another ISA adapter using a different brand of (low ESR) capacitors and it seems to work correctly!

FreddyVRetro, 2024-09-21:
Hi,

Thanks a lot for this information.

It may help other if you say more about the adapter model and capacitors.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

StolleMan, 2024-09-21:
Hello,

This is the board I used:
http://www.enide.net/webcms/index.php?page=ppc512-640-isa-expansion

I've attached the working and non-working BOMs. I haven't narrowed it down to a single component causing the dramas.

[image]

FreddyVRetro, 2024-09-23:
Hi,
It seems that you replaced the 10uF by 100uF ? (on both your designs)

StolleMan, 2024-09-23:
Looks like I followed the (incorrect?) BOM from the Tindie Store link: https://d3s5r33r268y59.cloudfront.net/boms/26932/2022-05-29-14-21-41/ams-ppc-isa-bom.txt

greukik, 2024-10-01:
Got the same issue using SN74HC244N ICs.
Tried to change +5V rail capas between 100uF, 10uF and 100nF wthout any change.
Will try Low ESR caps....

greukik, 2024-10-01:
> Looks like I followed the (incorrect?) BOM from the Tindie Store link: https://d3s5r33r268y59.cloudfront.net/boms/26932/2022-05-29-14-21-41/ams-ppc-isa-bom.txt
> 
> I'll try 10uF capacitors and see if there is any change

I've got the same issue with PicoMEM and xtcf. Xtide and adlib are running well.
But I'm lost with all your different BOMS and posts.
Could you please post exactly which capas you used because it seems to be confusion bw nF and uF when I read you. 🤔

StolleMan, 2024-10-01:
> I've got the same issue with PicoMEM and xtcf. Xtide and adlib are running well. But I'm lost with all your different BOMS and posts. Could you please post exactly which capas you used because it [...]

Here are some photos of the working board.

This BOM works for me:
https://d3s5r33r268y59.cloudfront.net/boms/26932/2022-05-29-14-21-41/ams-ppc-isa-bom.txt

There is a discrepancy in the filter caps between Enide's main page and the Tindie Store. In similar applications it seems that 100uf us usually used to filter. The silkscreen says 10uf but I definitely used 100uf

My adapter is working with or without low ESR caps. All I did was change the ICs to make it work.

[image]
[image]

JihemB, 2024-10-20:
Hello.
Yesterday I performed tests with PicoMem, my PPC640 and my brand new expansion board (Enide v2.1 version with 10 µF capacitors and SN74HC244N ics) : When my PicoMEM is connected on the farthest slot from the computer, I get a "parity error D000) almost 100% of the time, when connected on the middle slot it works about 50% of the time and eventually, when PicoMEM is on the slot closest to the PC it works almost 100% of the time (I switched off/on the computer about 50 times and it never failed again).

The question is: Is it matter of IC or capacitors, or both ?
SInce I have SN74HC244N, should I replace my 10µF capacitors with 100µF ones ?

My purpose is to add a Paradise VGA card too on the expansion board. Would it face the same problem or it is just a problem that is met only with PicoMEM ?

JihemB, 2026-05-09:
@l00nix 

Hi Captain Future/Capitaine Flam ;-)

As I wrote previously, the workaround I found is to put the PicoMem on the ISA slot that is closest to the computer. With this setup, I even can plug an ISA VGA card on the second slot and I never face the issue.

If this workaround doesn't work for you maybe you can change the 6 10µF capacitors  capacitors with 100µF which seems to be a good lead (but which I personnal didn't do yet). If this works for you please leave a feedback here.

Montie001, 2026-05-10:
> Thanks @JihemB <https://github.com/JihemB> for your answer.
> I have a PPC512 and a port board, but never tested yet.
> The behavious seems really random on PPC :-(

I never got it working tried every variation

Different BIOS too

:(