FreddyVRetro/ISA-PicoMEM #54: Feature Request: Raspberry Pi Pico 2

sdcoil, 2024-09-09:
Excellent job, Freddy!

Will you be adding support for the Pico 2 or the Pico 2 W (when/if it's released)?

Thanks

--Stacy

FreddyVRetro, 2024-09-09:
Hi,

The problem of the Pico 2 or Pico 2 W is that the feature that will boost the PicoMEM is not possible on them.
(The QSPI PSRAM support)
Then, there will be no advantage to use these boards.

I will check anyway for the Picoroni pico plus 2 or do a board with the RP3520B directly soldered.

So, Yes, but not with the pico boards.

sdcoil, 2024-09-09:
> Hi,
> 
> The problem of the Pico 2 or Pico 2 W is that the feature that will boost the PicoMEM is not possible on them. (The QSPI PSRAM support) Then, there will be no advantage to use these boards.
> 
> I will check anyway for the Pimoroni pico plus 2 or do a board with the RP3520B directly soldered.
> 
> So, Yes, but not with the pico boards.

Got it. Thank you for the quick reply.

hobopanda, 2024-09-13:
Hi, 

Looking at the new Pico 2 Data sheet I see that it supports External Quad-SPI flash just like the pico 1

FreddyVRetro, 2024-10-03:
I Close this one as Pico2 is of course planned :)

closed 2024-10-03 by FreddyVRetro

hobopanda, 2024-10-16:
The new by pico has 256K of extra onboard ram so potentially the 128K zero stay memory can be increased to 384K