FreddyVRetro/ISA-PicoMEM #57: No sign of life

raidersan, 2024-09-23:
I am on Pentium 133Mhz machine, pcchips motherboard (https://theretroweb.com/motherboards/s/pcchips-m530) with an AWARD BIOS and a VGA S3 Card (Genoa). I have tried all 3 firmwares with all combinations of IRQ and shifting the bios address from C800 to D000. I cannot find any sign of life of the picomem, I have never seen any PicoMem message at boot. 
Could it be the SD card? Formatted 64Gb on my Mac, it is readable by my PC as well.
Is there a way to tap into the PI's boot log to see what is going on?

raidersan, 2024-09-24:
Hi, I don't have a working floppy drive or disk. Is there any other way to tell? I see there is a header, can I plug in to that?

FreddyVRetro, 2024-09-24:
Hi,

Did you try the PREBOOT option in the config file ?

raidersan, 2024-09-24:
Yeah, no luck

raidersan, 2024-09-27:
[image]

Here is the debug output, looks fishy to me, surely those FF every second character should not there, right?
Also is the address 2a0 linked to an IRQ? I am on 3 at the moment but it is not increasing

lalcaraz, 2024-09-29:
Howdy. Same here. Received my board from Texelec today and I've been trying to get it to work for the last 2 hours on an IBM Model 25 (8086) and on a PentiumII 350MHz, still no luck.

lalcaraz, 2024-09-29:
Tried this morning with a different uSD card and no luck. Also tried formatting the card in Windows and no difference. I tried old firmware versions as well.

Could it be that either Texelec made a mistake when soldering or it is really incompatible with my Model 25 (8086)? I see the memory count, then the IBM "insert floppy" animation appears.  It could be that it needs to be at a different location in RAM to be executed. 

Finally, validated that the ISA slot was providing power by connecting the 5v/GND to a breadboard and measuring the voltage. All good. 

I will keep checking on this during the week.

lalcaraz, 2024-09-29:
Tried D800 as it was commented on an XTIDE post related to my machine and no luck. So far C800, D000 and D800 not working with my card. I’m starting to think it’s DOA.

raidersan, 2024-09-29:
Hi Luis,
can you replicate the issue I have while in debug? (see my screenshot above). You would have to adapt the command to look at the address space you have configured of course. I am no expert but I doubt the FF every second character in the PicoMEM BIOS is normal.

FreddyVRetro, 2024-09-29:
Hi Raidensan,

Thanks for the test and screenshot.

It is the port 2A3 that auto increment on read, not 2A0, 2A0 is the command status register and should be 0, it is probably 4 Because you wrote something on it.

If I see well, it is like if the Address 0 Bit does not work.
The A0 is shared with D0, so D0 should be stuck as well.

You can confirm if you do " i 2A1" or i "2A3", it should answer FF
Also, Do 
o 2A2 AA
i 2A2
o 2A2 55
i 2A2

This pin is the pin number 11, the 11th one from top right of the Pico:
[image]

You can check if it is soldered correctly

FreddyVRetro, 2024-09-29:
OOps, no, if A0 Does not work you will see the data in double, it is not the case here.

It really looks like a Timing problem, the PicoMEM is not fast enaugh for your PC.

Do you have the possibility to slow down the ISA BUS with parameters from the BIOS ?

FreddyVRetro, 2024-09-29:
> Tried D800 as it was commented on an XTIDE post related to my machine and no luck. So far C800, D000 and D800 not working with my card. I’m starting to think it’s DOA.
> 
> Edit: is there a way to debug what the Pico is trying to do while plugged into the retro computer? Or log to the uSD card what’s going on ?
> 
> Edit: source of the D800 address idea - https://forum.vcfed.org/index.php?threads/ibm-ps-2-model-25-with-xtide-universal-bios.1239707/
> 
> Edit: the Model 25 should be able to boot BIOS extensions as shown in this video. I don’t own an XTIDE so I can’t test it. https://youtu.be/1RFMgqRvYMw

Hi,

If the PicoMEM Green LED is on, the Pico initialized correctly.

I will definitively have to add LED Blink code for errors "Display".
A Firmware that display log on an emulated USB serial port is also possible.

If the PC Does not see the picoMEM, it can be due to lot of things, if it does not work on multiple PC, it is probablt Dead, I think you can return it to be tested again.

raidersan, 2024-09-29:
> OOps, no, if A0 Does not work you will see the data in double, it is not the case here.
> 
> It really looks like a Timing problem, the PicoMEM is not fast enaugh for your PC.
> 
> Do you have the possibility to slow down the ISA BUS with parameters from the BIOS ?

The bus is already running at its slowest at 8Mhz, don't think it can be slower than that and seems standard for ISA.
Can you tell me what PIN11 is connected to, I would like to do a continuity test on it. None of the pins look soldered very well in fact. 
a stuck AD0 makes a lot of sense for the symptoms I have, why do you think that is not the case anymore?

FreddyVRetro, 2024-09-29:
Hi,

Because if D0 is not connected, it read 2 times the same value.

Timing is not only the clock.  The biggest problem is the duration of the read signal.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

raidersan, 2024-09-30:
do you specifications for the read signal I can compare to with my oscilloscope?
Would a different bios help?

FreddyVRetro, 2024-09-30:
Hi,

I think I did not ask you if you tested the fast RAM firmware.

Were you able to test the I/O port 2A3h ? (Multiple in)

raidersan, 2024-09-30:
I have run the fast RAM firmware as well, not sure if that is the one I am on at the moment. Anyway I can check without reflashing?

-i 2A1 : always return 0
-i 2A3 : started at 80, then incremented by 1 every time it is run
-o 2A2 AA, -i 2A2  : returns AA         
-o 2A2 55, -i 2A2  : returns 55

lalcaraz, 2024-09-30:
I've requested an RMA from Texelec but I will do the tests you're suggesting.

raidersan, 2024-10-01:
@FreddyVRetro anything in the readings above that gives a clue?
Is there anything I can try with the BIOS, it's full of timing options
What about measuring the READ duration with my oscilloscope?

FreddyVRetro, 2024-10-01:
Hi,

Can you show me the available options ?
The one working is adding wait states.

"-i 2A1 : always return 0           > Should do it on port 2A0 (I suppose it is a typo)
-i 2A3 : started at 80, then incremented by 1 every time it is run
-o 2A2 AA, -i 2A2 : returns AA   > Should do it on port 2A0 (I suppose it is a typo)
-o 2A2 55, -i 2A2 : returns 55
"
What you show here as I/O is working perfectly, this confirm it is a timing issue for memory.
I/O Read and Write are longer, and the PicoMEM move IOCHRDY to have longer cycles.

In the next firmware, it will be possible to use the PicoMEM only in I/O to at least emulate the sound cards.
On the firmware you have, if the BIOS does not start, everything is disabled by default.
I am working on this in the extreme cases where the Memory emulation is not possible.

What fonction do you want to use ?

FreddyVRetro, 2024-10-01:
> I've requested an RMA from Texelec but I will do the tests you're suggesting.

Hi,
Thanks for your proposal, if you can do the same test as raidersan ?
If it answer to no I/O it is probably dead.

raidersan, 2024-10-01:
Unfortunately I wanted to use most functions, certainly sound, HDDs, network.
I tried to prove that A0 was going wrong but it looks fine, at least if I reverse engineered it right. Are you multiplexing bit 1 and 8 from the ISA to pin 11 on the pico? if that is right, I get continuity from the output of the mux to that pin.

Here are the BIOS screens I thought might be relevant, I have tried to slow down anything that made sense

[image]
[image]
[image]
[image]
[image]

lalcaraz, 2024-10-01:
Howdy. I reflashed PM_W_Jul16 and it booted on my P II. Saw the banner and the .CFG got created on the root of the SD.
However, on the Model 25 I don't see the banner BUT on Debug I can see the 10 on "i 2a0" and the increments on "i 2a3". 

This points to incompatible BIOS (Thanks IBM). Any idea how can we fix this? How can I be of help?

[image]

I can open a new ticket so we can focus on it.

raidersan, 2024-10-04:
@FreddyVRetro anything you can spot in my BIOS screenshots above that would help with timing to read the PicoMem BIOS properly?

FreddyVRetro, 2024-10-04:
Raidersan,

The only settings I see is ISA Clock.
Put the ISA Clock as low as possible, you may even try to reduce the machine main clock to see if it work.
(Not in the BIOS, I suppose with jumper on the motherboard)
Some BIOS allow to add wait states, but not this one :(

With the BIOS data read screenshot you did show, this clearly show a timing problem, but a Really tiny one.
The Fast RAM firmware should work, or if it does not, I still don't understand what is going on.

So, please, do all your test with the Fast RAM Firmware.

The PicoMEM will continue to progress, Its compatibility with system is growing.

Next Generation of PicoMEM will be more usefull for PC faster than XT. (Pico 2, faster bus....)
Even the current picomem will improve with the ability to use it even if emulated RAM does not work.
For example, Next firmware (This week) will allow for Audio emulation without BIOS.

FreddyVRetro, 2024-10-04:
> Edit: Scratch that. It's not booting anymore on the P II or showing on the Model 25. This is trificult.

Maybe the M25 corrupted the picomem config and it does not work on the PII anymore

raidersan, 2024-11-06:
Freddy, I have exhausted the possibilities of getting it to work with my Pentium I, it just does not like the timings of my motherboard. A bit strange that nobody else had the issue since I believe this Chinese generic product was sold in massive numbers under more famous (but still budget) names like PCChips. 
If you ever produced a new version of the picomem and would like to see if it solves this problem, let see know, I am happy to play the beta tester

FreddyVRetro, 2024-11-06:
@lalcaraz Hi, Just got a report of PicoMEM working on a model 30.

lalcaraz, 2024-11-06:
> @lalcaraz Hi, Just got a report of PicoMEM working on a model 30.

Thanks! 
I learned from an FB Model 25 enthusiast group that another Model 25 8086 owner made it work with a BIOS version 1.2A. I have v1 and since then I've been asking him to let me have a copy of his ROM to test but still working on that. Yet it works out of the box with the Compaq Portable. I'm coming to think this Model 25 wants to remain floppy-only until the rest of its days.

Max120l, 2024-12-19:
i've been having intermittent issues with a picomem 1.14 installed in at&t 6312 WGS. i'm assembling one of these computers from parts found online, so it's hooked up to a bench power supply for now. I was setting the voltage at 5v and it was ending up being 4.8 on the motherboard (which booted correctly). Under those conditions, the picomem showed no signs of life (including the led on the rp pico not lighting up. So I tried to raise the voltage to get 5v on the mother board and at that point the rp pico led lit up and it started working. I though that the raspberry pi pico had a buck converter to supply the 5v from whatever is fed into it... but it seems like to be faulty/not installed in this one. any thoughts?

Gilatrout, 2024-12-20:
Just installed a TexElec v1.14 board onto my Model 25.  No dice.  Just get the load floppy display.  I added the Preboot etc to my config.txt file.  Any suggestions appreciated.

Max120l, 2024-12-20:
I forgot to mention that I have the same board (TexElec v1.14). It looks to me like the d1 drops the voltage too much for the rp pico to work. I bypassed it and the voltage problem went away. I guess I'll just make sure not to plug the usb cable while the board is in the computer...

Gilatrout, 2024-12-20:
Thanks Max120I  I bypassed the diode on D1 with a bodge wire and the Pico lit up and is working like a charm.

So it seems that the problem is with the power

FreddyVRetro, 2025-01-31:
> i've been having intermittent issues with a picomem 1.14 installed in at&t 6312 WGS (286 12Mhz). i'm assembling one of these computers from parts found online, so it's hooked up to a bench power supply for now. I was setting the voltage at 5v and it was ending up being 4.8 on the motherboard (which booted correctly). Under those conditions, the picomem showed no signs of life (including the led on the rp pico not lighting up. So I tried to raise the voltage to get 5v on the mother board and at that point the rp pico led lit up and it started working. I though that the raspberry pi pico had a buck converter to supply the 5v from whatever is fed into it... but it seems like to be faulty/not installed in this one. any thoughts?

Hi,

The picoMEM use the U10 chip to re generate a RESET signal to the Pico,
This one has a limitted voltage range.
It was necessary to add it to have the picoMEM working on PC with a dirty 5V

julianfnunez, 2025-02-11:
@Gilatrout @FreddyVRetro Just another happy customer reporting that bypassing diode D1 brought my Picomem to life. Was about to send it back to Texelec. 

4.8v before the diode, 4.6 after the diode. With the diode on no signs of life after hours of debugging. Bypassing diode made the green light turn on and worked like a charm.

RetroBard, 2025-04-12:
I'm still struggling with mine. I managed to try the debug. At least I wasn't able to get any PicoMem related texts in the D000 address that I had set in the config. It is getting enough power, though.

[image]
[image]
[image]