FreddyVRetro/ISA-PicoMEM #58: Adlib only seems to work once, and won't work again until after a cold-boot

MacGyverr, 2024-09-28:
I have an IBM 5150, 64kb RAM, a 286 addin card, a XT-IDE (r629) at c800 with a Microtech microRAM addon attached bringing it up to 640kb and attached to an SD reader that I boot from at 300h, and a VGA card and IO card with floppy support plugged into a 360kb floppy and a Gotek running FlashFloppy.

My PicoMEM (firmware 2024-07-08) is set to d000 and IRQ 3, WiFi at 320 (I have also tried it at d800 with the same effect)
I use my PicoMEM just for the Adlib and WiFi so I have no HD or floppy images attached and it set to not use EMS or PS RAM, and to let XT-IDE boot.

The issue is when I cold boot from completely off, I can run Prince of Persia and hear music out the Adlib and effects out of the PC speaker perfectly.
If I Ctrl+Alt+Del and boot the same way again and run PoP, no Adlib music only PC speaker effects.

If I run something else from a cold boot that uses Adlib like Sim City Adlib works, but if I exit to DOS and try to run it again or PoP, no Adlib.

So the best way to describe the issue is that Adlib only seems to work one time per cold-boot.

Thank you for all your effort in this project, even Adlib support per session is still one more than I had before.
And the WiFi support is awesome. 

Let me know if you need me to perform any tests or give you any other information. 

[image]
[image]
[image]
[image]
[image]

FreddyVRetro, 2024-09-30:
Hi,

McGyver does not know how to fix something ? :)

I change a lot the audio part in the next firmware, I will try to reproduce the problem before releasing it.

FreddyVRetro, 2024-10-11:
Hi @MacGyverr 

Did you test with the latest firmware ?

MacGyverr, 2024-10-11:
I did, and it still does it.
I saw the new settings for CTS and Tandy as well as the address (I left it set to 388h)
Same thing, works perfect on a cold boot, but after a warm boot from Ctrl+Alt+Del no Adlib sound (aside from the PC speaker).
Before when it would warm boot it would keep playing the last tone that was playing, that has stopped, so this is better.
I can try to make a video if you want to see it. But it's just this: turn on, go to PoP folder, launch game, hear sound, CtrlAltDel, go to PoP folder, only PC speaker sound.

closed 2024-10-11 by MacGyverr