FreddyVRetro/ISA-PicoMEM #59: external bios checksum incorrect

thecaver99, 2024-09-29:
i have an amstrad ppc520 and plugged the picomem in to the isa expansion adapter and i get the following error external bios checksum incorrect D0000,
if i have the boot code on and an xt-ide installed along side the picomem boots and works great but it wont work stand alon.
if i can supply any information with regards to helping resolve this i am only to happy to help.

FreddyVRetro, 2024-09-29:
Hi,

You can check the other subject here, with a PPC640.
It is a problem of 5V Filtering, the capacitors on the adapter are not sufficient or not with the correct quality.
(100uF Low ESR is preferred)

https://github.com/FreddyVRetro/ISA-PicoMEM/issues/53

thecaver99, 2024-09-29:
thanks FreddyVRetro,

will check this out, as with the xt-ide in place its working perfectly, going to order some 100uf low esr Panasonic caps for it soon, will let you know how it goes, as if it all goes well i will be ordering a second one for the ppc as this one is from my pc1640.

StolleMan, 2024-09-30:
@thecaver99 

In the other thread referenced, I assumed the issue was with the filtering capacitors. The "fix" was to change the  buffer ICs. Which ICs did you use on your adapter?

FreddyVRetro, 2024-09-30:
Hi,

No, the fix was to change the capacitor.

Can you tell me more about the XTIDE ?
Model ( Capacitor used to filter the 5V ?)

As the solution is probably to add this filter capacitor to the PicoMEM

thecaver99, 2024-09-30:
> Hi,
> 
> No, the fix was to change the capacitor.
> 
> Can you tell me more about the XTIDE ? Model ( Capacitor used to filter the 5V ?)
> 
> As the solution is probably to add this filter capacitor to the PicoMEM

I will take a pic of the xt-ide later when im home,
but there is another twist, i have a generic bnc nic and fitted an xt-ide bios to that, when the ppc boots it compains about the nic rom having the wrong checksum now and the picomem works without issue.
i will also take a pic of that at the same time as my xt-ide.

thecaver99, 2024-09-30:
for the xt-ide, the following link is the one i have.
https://www.tindie.com/products/glitchwrks/glitch-works-xt-ide-rev-4b/

StolleMan, 2024-09-30:
> Hi,
> 
> No, the fix was to change the capacitor.
> 
> Can you tell me more about the XTIDE ? Model ( Capacitor used to filter the 5V ?)
> 
> As the solution is probably to add this filter capacitor to the PicoMEM

LOL, if you say so... I'm pretty sure I know what I did though...

thecaver99, 2024-09-30:
> > Hi,
> > No, the fix was to change the capacitor.
> > Can you tell me more about the XTIDE ? Model ( Capacitor used to filter the 5V ?)
> > As the solution is probably to add this filter capacitor to the PicoMEM
> 
> LOL, if you say so... I'm pretty sure I know what I did though...

so what is the differance to the reset, as looking the only differance is the low esr caps, disc caps 100nf, and idividual resistors for the rn?, im just curious as trying to make the board as small as possible to fit inside the ppc case.

StolleMan, 2024-09-30:
On one board I used CD74HC244E ICs. This is supposed to be a direct replacement for SN74HC244N. It would work fine for XT-IDE but not picomem. When troubleshooting I tried different combinations of capacitors which made no difference. The only thing that 'fixed' the problem was swapping the ICs for SN74HC244N.

FreddyVRetro, 2024-10-11:
@StolleMan I am coming back here late :)

Good to know, but even SN74HC244N is not a good idea, as TTL Compatible components are HCT ones, not HC.

FreddyVRetro, 2025-05-13:
> I will take a pic of the xt-ide later when im home, but there is another twist, i have a generic bnc nic and fitted an xt-ide bios to that, when the ppc boots it compains about the nic rom having [...]

Hi,

Did you finally fix the problem with added capacitor ?

thecaver99, 2025-06-22:
i have not been able to fix it as yet.
i know im going to ask an odd question but would it be possible to make one of these that would be compatible with the pc20/pc 200 machines that would be longer in length and shorter in height as there is only about 5cm head room?