FreddyVRetro/ISA-PicoMEM #62: Doesn't work on M396F V2.7 (SARC RC2016A)

andreacampanella, 2024-10-07:
I cannot make the PicoMEM working on [this motherboard](https://theretroweb.com/motherboards/s/pcchips-m396f-v2.7). 

I tried to swap bios a dozen time, the problem is that this board doesn't allow to change the ISA timing as it was very cheap.
it's a shame as there is an opesource project about it. 

the led on the PicoMEM turns on, but nothing else happens, this is with the last bios

FreddyVRetro, 2024-10-07:
Hi,

286 and 386 played with the ISA timing and are too old to have ISA Timing settings...
Maybe try to reduce the clock ? (Jumper on the board)

andreacampanella, 2024-10-08:
No, I went down as much as I can with no luck. 

I would like to collect some informations about it, can you tell me which lines I can scope to figure out what's going on?

FreddyVRetro, 2024-10-11:
Hi,

Do you have something to booth the PC with ?
You can test if it is detected with I/O via PMINIT and enable sound card emulation now.
also use the DEBUG "D" command to check the BIOS
Doing d D000:0000 if setup as default.

Also, ROM Shadow need to be disabled.

Regarding the signals, no, this is not as simple as this, I need the PC with the PicoMEM on it and logic analyzer to see what is Ok or not.

andreacampanella, 2024-10-11:
Yeah, I can boot it from a SCSI card, I will try as soon as I can :)

FreddyVRetro, 2024-10-14:
Hi,

If it is timing related and the fast RAM firmware does not work, there is low chance.
I am not sure I can speed up the code now, it is already 100% in ARM assembly.

I plan to change the multiplexing in the Next "PicoMEM 2" board, in some months.