FreddyVRetro/ISA-PicoMEM #64: Why has the support for small capacity MicroSDs been removed?

birnwieapfel, 2024-10-15:
Hello,

I just updated my PicoMEM and wondered why my 2GB MicroSD card wasn't detected anymore. Then I saw the following in the firmware release notes: "Removed : It does support only SDHC (No more small capacity MicroSD)"
I was really happy to be able to use my "old" 2GB MicroSD cards. In a typical ISA system, there is normally no need for larger disc capacities.
What was the reason to remove the support? Is it possible to reactivate the support (e.g. in a separate firmware)?

Regards

Robert

FreddyVRetro, 2024-10-15:
Hi,

I can understand it is frustrating :(
In fact it is not me that removed the support

I used this library that drop these SD support:
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico

And it was mandatory for me to include the last build to be able to use the PicoSDK 2

birnwieapfel, 2024-10-15:
Ok, understood. Thanks for the explanation.

closed 2024-10-15 by birnwieapfel