FreddyVRetro/ISA-PicoMEM #65: DHCP doesn't work

megatog615, 2024-10-26:
I have `pm2000.com 0x60` loading on startup conforming to the FreeDOS fdnet network card setup, using fdnetpd.bat to load pm2000. pm2000 reports success and gives me a MAC address and other information. However, dhcp.exe reports that the DHCP requests are timing out.

Is there a way to troubleshoot this with more verbose output?

FreddyVRetro, 2024-11-14:
Hi,
I asked mtcp author to help you if possible.

mbbrutman, 2024-11-15:
mTCP has a fairly extensive trace facility built into it.  Please see the section on "Debugging" in the mTCP PDF documentation.  In short, set DEBUGGING=255 and LOGFILE=trace.txt, run DHCP, and then post the trace.txt file.

megatog615, 2025-01-03:
Hi. Sorry for the late response. I have a trace.txt but after reviewing the contents of it I'd rather not share it as it seems to contain identifying information such as my public ip address from packet responses. It's probably not useful to anyone but I'd rather it not be on the internet.

Yes, there are packets received in this log, mostly from some of the smart devices around my house that belong to my housemates (😭). What info can I provide from the log that would be helpful?

mbbrutman, 2025-01-03:
You can send it to me directly as per the mTCP instructions.  My email generally doesn't get forwarded to the public internet.

megatog615, 2025-01-03:
Okay. I sent it to you.

mbbrutman, 2025-01-04:
I think we found it.  The short story:

- mTCP is sending a packet with an odd number of bytes.
- The packet driver detecting a 186 or better CPU which can use REP IN/OUT instructions.
- The packet driver is going to use those instructions to do a REP OUTSW. 
- The packet driver shifts the length in CX to the right one bit (divide by two) to compute the number of words for REP OUTSW
- Not enough data is transmitted.

The immediate error is the shift to divide the length by two.  With an odd number of bytes that winds up throwing a byte away, and that's probably what is causing the packet driver to report the error back to mTCP.  (It must be an underflow condition in the driver or hardware.)

Luckily the 186 detection code does not detect a V20, so V20s are using the older, slower byte-at-a time path.

Anyway, I have some mTCP test code that never tries to send an odd-length packet and two people have confirmed that fixes the problem.  I'm not posting that link because it's debug code, and I expect we'll just get the PM2000 driver fixed.

FreddyVRetro, 2025-01-14:
Hi,
Were you able to fix your issue ?

mbbrutman, 2025-01-15:
I used the traces sent to me directly to diagnose the problem so this bug should be resolved, but it would be good to confirm it directly.

megatog615, 2025-01-16:
Sorry for the delay. Yes, the new PM2000.COM fixes the problem. I think this can be closed.

closed 2025-01-18 by megatog615