FreddyVRetro/ISA-PicoMEM #66: PCB conversion for pc/104

Hotless80, 2024-11-03:
Hi, I would be very happy if a pc/104 version of PicoMEM was created, or if you gave me the opportunity to create it, I have already converted 2 other cards to pc/104 standard, including PicoGUS, but PicoMEM would be much more useful for me

Aisflou, 2024-11-04:
Where are those converted boards? pc104 its a fascinating platform :D

Hotless80, 2024-11-04:
[image]
[image]

Hotless80, 2024-11-04:
Just a common sound card.

A retrocomputer like this?
<details>
  <summary>Spoiler warning</summary>
  
[image]
[image]
  
</details>

Aisflou, 2024-11-04:
Can you share the info about the sound card? thats usually whats more needed in these computers! :D Yeah, that case lloks good indeed! im more of the desktop kind, but that one lloks reall good also! :D more info about it? hehehe Nice work!

Hotless80, 2024-11-04:
As soon as the project is improved I will publish it (it's a bit noisy), but now I think we're going too far off topic

Aisflou, 2024-11-04:
True, sorry! :(

EffCancer, 2025-01-06:
I just saw this as I was pondering doing the same thing. I created a protoboard PC/104>ISA interposer to accomplish this task, but wanted to go even smaller for my 486 SBC micro-retro computer, so I ended up here. I hope you manage to find the time to put this together, but in the interim, I hope you won't mind if I have a go at it.