FreddyVRetro/ISA-PicoMEM #67: Problem with Logitech F310 Joystick and additional questions

themustysmell, 2024-11-05:
Hello.
My PicoMEM arrived last weekend. Board revision 1.20A with the BIOS from October 3, 2024. The card is installed in an IBM PS/2 Model 30 (8086) together with a VGA card and an XT-IDE. The operating system is MS-DOS 5.0 and Windows 3.0. So far everything works very well, but I can't get any further with the joystick. My newly purchased Logitech F310 joystick doesn't work in either D or X mode. It is recognized by the PicoMEM. But programs such as MSD.EXE show no reaction when the buttons are pressed. Games can't find it either. USB and joystick are switched on in the BIOS. PMINIT.EXE is started automatically with the parameter /J 1 after booting.
Does anyone have any ideas what else I could do or test? And I would be very interested to know which joysticks (manufacturer and type) have already been successfully put into operation by the community.
Many thanks.
Greetings from near Hameln (Germany),
Marco

FreddyVRetro, 2024-11-14:
Hi,

When you boot from the PicoMEM with the joystick enabled; the PicoMEM Will display what it detect.
What is it telling ?

I use another librady that is named XInput, so I don't know it well enaugh to tell you if your joystick is supposed to work or not.

FreddyVRetro, 2024-12-03:
Hi,
I am thinking about something.
If there is already a joystick port on your machine, it will not work.
(Like Sound blaster card)

The joystick interface need to be disabled.

themustysmell, 2024-12-03:
Hello Freddy.

Apart from the PicoMEM with the AdLib module, there is no other sound card installed. Not even on the mainboard.
I suspect the problem is somewhat different. If I turn off USB and joystick in the PicoMEM BIOS and start MS DOS 5.0, MSD.EXE still finds a game port. Perhaps another device is using the I/O addresses 0x200 - 0x20F?
My question now would be, how can I check whether these I/O addresses are used?
The PC used here (IBM PS/2 Model 30) is special anyway and I wouldn't be surprised if another device is using these I/O addresses.

themustysmell, 2025-01-20:
Hello FreddyV. After installing the latest firmware (PM_W_Jan1.uf2) for the PicoMEM today, one of my two gamepads is recognized. The Logitech F310 still isn't. But the one from Retoo (ASIN: B0CCVW36KJ, model E243) is. Strangely, MSD.EXE doesn't recognize either of them. But I can do the joystick test with Checkit3 without any problems. Wolf3Dc can now also be controlled with the Retoo gamepad. I'm attaching a picture of the BIOS message. This ticket can be closed.

[image]

closed 2025-01-20 by themustysmell