FreddyVRetro/ISA-PicoMEM #68: Ram parity error - Goupil G5 80286

Cocolas44, 2024-11-09:
Hello,
I have a "Ram parity error" when I try to boot a OS, I have disable the emulated memory but the problem continues:
[image]
I have tried PREBOOT, BIOS C800 even TANDY option. Only BIOS 0 work :(

Is there a way for me to activate a debug mode to find a solution?

FreddyVRetro, 2024-11-14:
Hi,

The debug firmware I have can't track a problem like this.
I see from the screenshot you also try with the FAST RAM firmware.

Maybe you tried, but is it able to BOOT via something else than the PicoMEM ?

Cocolas44, 2024-11-17:
Hello,
Yes, I have tried many Bios versions, only the "PM_W_Mar17_C8_DU" do not boot.
If I boot from floppy I have the same error but if I boot from  internal hard disk (FreeDOS) I do not have ram parity error. From the OS I have tried to reboot (ctrl +alt + supp) then I do not have access to Picomem BIOS :(
I also have tried to access to the LAN with PM2000.COM but it doesn't detect it.

FreddyVRetro, 2024-11-17:
You may try to delete picomem.cfg

Cocolas44, 2024-11-30:
Hello,
I have tried a lot today:
Delete picomem.cfg -> Help with EMS, now it work, thanks! :) But if I try to boot on emulated HDD I still have RAM parity error. Sometime I have this error under PicoMEM Bios.
Older version -> always the same parity error :(
Activate PM RAM -> I have a write error on SD card and Picomem hang :( I have to delete cfg file.

Cocolas44, 2024-11-30:
Something new, if I reboot from dos I have "Config file write error"