FreddyVRetro/ISA-PicoMEM #69: Schneider EuroPC1 PicoMem initialization or settings option not shown on boot

RetroBard, 2024-11-09:
I'm trying to get the PicoMem (rev 1.20 from SerdaShop) settings available on my Schneider EuroPC1. I have the latest Fast RAM firmware installed. PicoMem is powered up as the green light is on, but the PicoMem is not detected by the system after RAM check nor is the option to enter the PicoMem shown. I'm running the EuroPC at 9.54 MHz, but I tried also with 7.15 MHz.

Anything Else I could try?

FreddyVRetro, 2024-11-14:
Hi,

EuroPC1 was  difficult to have working.

Maybe some electrical problem on some machine add more constraints on timing.

Next PicoMEM Firmware will be a little faster, it may be better for this case.

Did you try by plugging the PicoMEM Via USB on another PC ?
(To have 2 Power source)

RetroBard, 2024-11-15:
Thanks for the response! I tried plugging an external power source to the  USB port, but that made no difference.

FreddyVRetro, 2024-11-15:
Did you test it on another PC ?

RetroBard, 2024-11-16:
Yes, I tried it on my 386SX and it’s working there.

RetroBard, 2024-12-01:
I’ll be waiting for the new version of the PicoMem firmware to come out.

couchgott, 2024-12-03:
Hi folks, 

just mentioning i do have similar Issues with an Schneider Euro PC II. 
tested several settings but i do have always the same behaviour. 

without USB Port connected to external power or with connection: 
9Mhz -> PicoMem recordnized. I end up with "Start legacy boot strap" 
7,x;Mhz -> same
4,x Mhz -> Error: ROM Checksum error D000
I disabled every RAM / Audio setting. -> same behaviour
I created a fresh img for the HDD -> Same 

After some Seconds, the PicoMem gets into a loop and tries again to boot, but again runs into timeout and loops.

FreddyVRetro, 2024-12-03:
Hi,
Start Legacy Boot strap means that you have the Disable PicoMEM boot option selected.

At 4MHz, the EuroPC2 ISA is probably faster, need the Fast RAM PicoMEM Firmware.

closed 2025-01-14 by FreddyVRetro

RetroBard, 2025-01-15:
I tried the latest fastram firmware, but still not getting the boot options in my EuroPC1. Mine has the NEC V20 CPU, if that’s of any help.