FreddyVRetro/ISA-PicoMEM #70: NE2000 (PicoMEM) in combination with LSL and IPXODI – is that possible?

themustysmell, 2024-11-11:
Today I executed the following commands under MS DOS 5 on an IBM PS/2 Model 30 (8086):

LSL.COM
NE2000.COM 0x60 3 0x320
IPXODI.COM

The NE2000.COM is the one from here: 
https://github.com/FreddyVRetro/ISA-PicoMEM/tree/main/drivers

My goal is to connect two computers via IPX (for network-capable games). The NE2000 emulation of my PicoMEM works (tested with mTCP and httpserv).

My question would be – is what I am planning even possible? In relation to the interaction of LSL, IPXODI and PicoMEM/NE2000.

The result of my attempt can be seen in the following picture.

[image]

Greetings,
Marco

FreddyVRetro, 2024-11-14:
Hi,

I am not a network expert to tell you if IPX is possible via the NE2000 Packet driver.
But I think another driver is needed.

cml37, 2024-11-14:
Yea, you are going to want to use a Novell NE2000 ODI driver as opposed to a packet driver, it's a little confusing since it will also be a .com file.

mbbrutman, 2024-11-15:
Hi ...  use one set of drivers on an Ethernet card at a time.  Trying to run a packet driver at the same time as a different driver on the same card is just going to lead to problems or crashes.

You can't have two different competing pieces of software touching the same hardware.  It doesn't work.  Load one or the other.  If you must have both types of drivers you will need a second network card.

themustysmell, 2024-11-15:
Hello.
Thank you for the feedback and tips. After further research, I am now pursuing the following approach:

PM2000.COM 0x60
LSL.COM
PDETHER.EXE
IPXODI.COM

With a suitably adapted NET.CFG. At least IPXODI can now be loaded without errors.

Next, I need to configure a second computer accordingly (a self-built 286) and bring it into the network.

As I wrote in the first post, I want to play network-capable games via IPX. Specifically, it is about this game: ([Multiplayer IPX for DOS: Elephant Herding](https://thandor.net/article/36)). This supports IPX and also runs on an 8086 processor with 640kB RAM.

If I manage to network the two computers with this game, I will submit the corresponding NET.CFG.

Regards,
Marco

themustysmell, 2024-12-03:
I'll try to answer the question myself. After reading the previous answers and after a few more attempts, it is as mbbrutman wrote above, that you have to decide. Either I communicate via TCP/IP and use a packet driver for this (for PicoMEM, the PM2000.exe, for example). Then I can use the programs in the MTCP Suite, for example.
Or, as in my case, to use the IPX protocol, you need an MLI driver (MLID - "Multiple Link Interface Driver"). In this case, the loading order would be:
lsl.com
MLID
ipxodi.com

But there is no such MLI driver for the PicoMEM. I would be very happy if someone could confirm this.

As a workaround, I currently have an Accton Ethernet Pocketadapter LPT running on my PC with suitable MLI drivers and also suitable packet drivers.

closed 2025-01-20 by themustysmell

cml37, 2025-02-13:
Heads up that recent PicoMEM builds seem to have problems with the NE2000 LSL drivers.  See here: https://github.com/FreddyVRetro/ISA-PicoMEM/issues/91