FreddyVRetro/ISA-PicoMEM #71: Support for Tandy 1100FD possible?

hexbus, 2024-11-14:
Do you think you might be able to support a PCB for the Tandy 1100FD (an XT class system) through an ISA adapter card?

It has what appears to be all the pins for an ISA bus - 

[image]

It needs this connector though (we'd have to identify it):

[image]

But it has quite a bit of room (lower right):

[image]

I'd be willing to help since I have the Tandy 1100FD and associated Panasonic BP150 units (they're the same thing essentially).

Here's pics of the only expansion card (1M EMS) that they ever made for it:

[https://imgur.com/gallery/panasonic-bp150-cf-ba161-1mb-ram-card-yOlTTvy](url)

FreddyVRetro, 2024-11-16:
Hi,

It should be possible.
I plan to work on a Tandy HX/EX version next year.

FreddyVRetro, 2024-11-16:
I know it is different than EX/HW :)

My time is limitted, currently working on another machine PicoMEM Version (Amstrad CPC)
Also working to prepare Pico2 based PicoMEM ;)

hexbus, 2024-11-16:
I’ll see if I can get someone in my local group to help with the PCB or adapter if you need assistance.   We can product prototypes and help you test for you if you like to make it faster.  Here is a group member familiar with PCB layout and what they produced: https://github.com/V2RetroComputing/analog

FreddyVRetro, 2025-01-14:
Hi,
Did you progress?