FreddyVRetro/ISA-PicoMEM #72: I/O Port Failure - no tandy music / emulation.

user5518, 2024-11-14:
My PicoMem is in a Toshiba T3200 with a 286 12MHz and 1MB RAM (and a ET3000AX with 512KB)  installed, primary for Music emulation.

Adlib and CMS are working fine. But Tandy isn't (anymore). 

For some reason, I sometimes (not always) get an "I/O Port Failure" and the Tandy music in "The Secret of Monkey Island" does not play. I initialized the card with "pminit /tdy 1".

(I started the game with "monkey ts" (no sound), "monkey a" (for adlib) "monkey g" (for cms) oder "monkey i" for the internal speaker are all working fine.)

[image]

What can I do to make Tandy work again?

user5518, 2024-11-14:
No, not so far. Let me try that, I will report back. Thank you.

FreddyVRetro, 2024-11-14:
Hi,

The Port failure is normal, it I did not found why PMinit sometimes fail on it, but the port are working.
I checked the PMinit and I forgot to code the tandy initialisation in it :(

Did you try to enable it with the BIOS ?

FreddyVRetro, 2024-11-14:
Hi again,

On a 286, you can't use the Tandy audio with the normal port, as it is C0 and this is the port for the 2ND DMA Controller

To have Tandy playing on Monkey island, you need to use this default port, the /tdy option configure the port 2C0 and need to have a modified monkey island executable.

You can find infos on all this on vogons forum, posts about the Tandy Board

user5518, 2024-11-14:
Thank you for your response. I was just try out different ports, and the last one (0C0) is working fine. :)

I was pretty sure 1E0 was working before I replaced the fans in this unit, but now I'm happy, that the tandy music is playing again.

[image]

Now I can try out Wifi, the Harddisk and RAM-feature.

And of course: Thank you very much for your hard work on the PicoMEM, it is a really great piece of hardware.

closed 2024-11-14 by user5518