FreddyVRetro/ISA-PicoMEM #73: PicoMem Audo Edition No Bootscreen on Schneider Euro PC 1

Peabrain, 2024-11-15:
Hello,

i bought a PicoMem Audio Edition from Serdaco and it doesn't work. I copied the Firmware via USB connection, but the connection never closes the window automatically. i put in in my schneider, but it doesn't work. i also but it in my commodore pc-10 iii, but it still doesn't work. I have also a XT-IDE card. this card works in both system. 

Best Regards

FreddyVRetro, 2024-11-15:
Hi,
We can say that you are not lucky.

Both commodore pc10 and europc1 were the more difficult to have working due to their fast timing. 

I did the fast ram firmware for them.
So, you need to change the firmware for them. 

I don't understand what is confusing.
I wrote that only the picomem 1.1 need to have the PC powered on.

The picomem is programmed like any pico.
While you press the boot button, plug the usb.
Then copy the file.

Peabrain, 2024-11-16:
thank you for the fast answer.

"The picomem is programmed like any pico. While you press the boot button, plug the usb. Then copy the file."
this is what i have done. but i dont know, what exactly shut happen. the description says  "wait until the folder automatically closes and it is finished.". but the folder doesnt closes. and you mean a windows pc, not the retro, right, for transfering the data?
what should it looks like on the pi (folder)?
please help. i like the projekt and it matches so  nice to the euro pc :-)

Peabrain, 2024-11-16:
yes i did it. and there are 2 files. index.htm and info_uf2.txt

in info_uf2.txt =>
UF2 Bootloader v3.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2

and then i copy PM_W_Okt5.uf2 on the device.
now the are 3 files in the RPI-RP2 device =>
index.htm
info_uf2.txt
PM_W_Oct5.uf2

and then, what should i to now? wiat? should i get the device out and put it in the euro pc? did the pico turn off itself?
the rpi did nothing.  do i have to rename the n.uf2?
the rpi's led is also turned off. how can i see, that the files is correct installed?

best regards

Peabrain, 2024-11-16:
i got to manage to copy the .uf2 file and the rpi turned off the device. it seems to copy the new firmware to the rpi. but it did work in the euro pc. did i have to copy something on the sd-card?

FreddyVRetro, 2024-11-16:
Hi,

The firmware to use with your machines is the Fast RAM One :
[PM_W_Oct5_FR.uf2](https://github.com/FreddyVRetro/ISA-PicoMEM/blob/main/firmware/PM_W_Oct5_FR.uf2)

When you say it work, the PicoMEM Work now ?

Peabrain, 2024-11-16:
ok, it worked one time. but i can't remember what i've done to make it work.

closed 2024-11-16 by Peabrain