FreddyVRetro/ISA-PicoMEM #77: SCrewed up menu on Headstart Explorer

AndrewBSSC4, 2024-12-03:
Hi. Trying my Picomem on a Headstart Explorer and the menu comes up all screwed up.  I had to use the "PREBOOT"  option so that Picomem startup would pause and wait for the "S" option to enter the menu.  But it seems it doesn't like the character set or something.   This system is suppose to have a CGA, MDA, Hercules compatible video card.  I also tried it on a real CGA monitor and it's the same so it's not a problem with the video conversion.  Are there any options for the config.txt to change the characters set used?  Thanks for this great project.

[image]

AndrewBSSC4, 2024-12-08:
It uses the Yamaha V-6366C-F chip.  The datasheet is available on line if it helps: 

 https://4donline.ihs.com/images/VipMasterIC/IC/YAMA/YAMAS00459/YAMAS00459-1.pdf?hkey=FB3F1F3F2A09A989A6BF9D772C3B8264

AndrewBSSC4, 2024-12-17:
Just a follow up. for the Headstart Explorer. If you setup the disk functionality on another system it will work in the Explorer.  As well as the sound functionality.  So it appears only the memory extensions don't work.  You will need to run with the "PREBOOT" in the config file.

FreddyVRetro, 2024-12-20:
> Just a follow up. for the Headstart Explorer. If you setup the disk functionality on another system it will work in the Explorer. As well as the sound functionality. So it appears only the memory extensions don't work. You will need to run with the "PREBOOT" in the config file.

Hi,

Nice.
Yes, the PicoMEM reset the memory settings if a difference is found between the saved RAM map and hardware detected RAM.
I may have to work on a way to force RAM settings in the config file, in case the BIOS menu does not work like on your PC

AndrewBSSC4, 2024-12-20:
I was actually thinking of that.  If the config file could be expanded to allow all or almost all settings to be manually programmed in the config.

Great project.

FreddyVRetro, 2024-12-20:
> I was actually thinking of that. If the config file could be expanded to allow all or almost all settings to be manually programmed in the config.
> 
> Great project.

It need tests, because the BIOS also act independently from the Pico, so it will continue to RESET the settings, need to have the option, plus disable the config reset and test if the new config is valid. not that simple

You have a weired machine, as the PicoMEM BIOS use the PC Text display BIOS fonctions.
It is not really PC Compatible.