FreddyVRetro/ISA-PicoMEM #78: Wifi Networking IRQ issue

gdattilo, 2024-12-20:
I'm using firmware PM_W_Oct5 on a book 8088. Something strange seems to be happening with the Wifi interrupt. The interface only works when I use the same interrupt for the interface that the PicoMem is using. It doesn't seem to matter how I set the interrupt in PicoMem(PM) setup. I can only pull an IP address if I issue the NE2000 command with the same IRQ as that of the jumper on the PM board. The port setting in PM setup seems to work as expected. As long as I issue the NE2000 command with the port as it is set in PM setup, the proper MAC is reported. I must issue the NE2000 command with the IRQ argument the same as the jumper or Brutman DHCP fails and says there is no activity on the wire. 

I checked most if not all of the variations PM setup Wifi IRQ, IRQ jumper position and the IRQ argument to NE2000. Best I can tell the interface works just fine as long as NE2000 is issued with the same IRQ as the IRQ jumper position regardless of the setting in PM setup.

I didn't see an other issue that seems like this one. If I'm double reporting a known issue, I apologize. This is my first report.

gdattilo, 2024-12-20:
Now that I look a little closer, this might be same think that was happening in issue 65 from October. That issue was framed as a problem pulling and IP address with Brutman DHCP. I think there is a good chance this is the same problem.

FreddyVRetro, 2024-12-20:
Hi,

The PicoMEM has code in its BIOS to use the Interrupt as a multiplexer.
It can ask to generate any Interrupt.

But maybe this interrupt code is redirected by something else, then does not work anymore (And this force you to use the same IRQ for the PicoMEM and ne2000)

If you use PM2000, it will force again to redirect the interrupt to the PicoMEM code.

gdattilo, 2024-12-21:
Hi Freddy,

Well I could have sworn that I tried PM2000.com with the same results as I described above but apparently I'm wrong. I did as you suggested and replacing NE2000 with PM2000 and now Wifi is working with an interrupt different from the PM itself. Sorry raising this issue when it seems it was not really an issue.

While we are messaging, I'd like to take the opportunity to thank you for creating the PicoMem. This thing is super cool. Keep up the good work. It is really appreciated.

One more thing. You mention multiplexing of the IRQ through the PM. Is there any reading available that discusses how this work. This sounds very interesting. I appreciate that the source for the PM bios is not open but if there is any public reading available I'd love to take a look.

I think you should consider this issue closed. Thank you once again.

Gus

FreddyVRetro, 2025-01-05:
Solved

closed 2025-01-05 by FreddyVRetro