FreddyVRetro/ISA-PicoMEM #80: At&t 6312 WGS / Olivetti M280 - EMS available intermittently and buggy floppy img menu

Max120l, 2025-01-05:
[image]
[image]

On a 286 system. I was able to turn on the ems when i first installed the picomem and it worked perfectly for a few days, then it stopped being added  and I wasn't able to select it anymore. Tried all the firmware versions to no avail. Any suggestions?

FreddyVRetro, 2025-01-05:
Hi,
Yes, delete the picomem. Cgf file
It seems corrupted.

Max120l, 2025-01-05:
I tried that and didn't change anything. I can upload/send you a copy of the .cfg file maybe?

Max120l, 2025-01-06:
The disk names are now working fine after removing the cfg file. The ram test during startup fails,  and here is a screenshot of when i enable PM ram
[image]
[image]

Max120l, 2025-01-07:
[image]
Running pm test i get some errors too.

FreddyVRetro, 2025-01-07:
Hi,

Of course, there is no RAM at these position.
PMTEST is hardcoded to a configuration.

MicroSD and PSRAM share the same bus
If there is a problem on the MicroSD, PSRAM may fail.

Can you see if the PSRAM error is still here without MicroSD, then change the MicroSD ?

closed 2025-01-14 by FreddyVRetro