FreddyVRetro/ISA-PicoMEM #84: I am getting memory parity non-maskabe interrupt at different addresses

nikolarun, 2025-01-18:
Hi Freddy,
I am getting those errors when I run dir, if i say S  , all is good. this only happens once in a session.

[image]

Is this my xt memory going bonkers or the pico ?

FreddyVRetro, 2025-01-31:
Hi,

What is your PC model ?
Did you try with the Fast RAM firmware version ?

nikolarun, 2025-01-31:
Hi Freddy, I will try the fast firmware. 
My pc is an XT clone with NEC V40 maed by a Korean company SBC. 
You helped me over on the VCFED forums with getting the expanded memory working by downlocking the CPU to 4.7mhz.

nikolarun, 2025-03-24:
Hi Freddy, I just loaded the latest FR firmware, but  I still get the issue. Any other ideas ?

nikolarun, 2025-05-27:
I will give the new FW a go, will report if it fixes the issue