FreddyVRetro/ISA-PicoMEM #87: Max disk images is too low

megatog615, 2025-01-20:
I noticed that some floppy disk images were not appearing in the list after I added some today. I checked the code and there is in fact a hard limit of 32 images.
https://github.com/FreddyVRetro/ISA-PicoMEM/blob/d73e0bb30156d366e7754409b3ba893451bd13b0/src/pm_disk/pm_disk.cpp#L41

Is it really necessary to cap the limit at 32 or is there a more technical reason?

FreddyVRetro, 2025-01-22:
Hi,

The limit can be pushed more, it is just that I found this number "Correct"

It is limitted in the way (Where in the RAM) it pass the image list to the BIOS.

I will think about it for next firmware.

FreddyVRetro, 2025-03-11:
Hi,

Pushed to 128 on the next firmware.